morrownr/88x2bu-20210702 #82: Device not appearing

kurahaupo, 2022-09-23:
I need to make this work with an older 4.4 kernel, and I'm up for patching this rtl88x2bu driver if necessary, but I could do with a few clues on where to start.

I've compiled the module and successfully used `insmod` (& `rmmod`) on it, but no (new) devices are presented on `ip link` or `iw dev`, even though it shows up correctly in `dmesg` and `lsusb`. (Until I can get that far, it seems pointless trying to configure

I've even tried `insmod`, insert device, `rmmod`, remove device, insert device, `insmod`, but no change.

Are there any command line parameters I could add to `insmod` to help find out what's going wrong?

This is a Dell d630 Latitude with a T-7100 2-core CPU and 8 GiB RAM; it has built-in ethernet (which I use most of the time), and WiFi (which is turned off by a hardware kill switch, but is still the only thing that shows up in the `iw` commands below).
```
$ uname -mr
4.4.0-140-generic x86_64
$ mokutil --sb-state
bash: mokutil: command not found
$ rfkill list all
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: yes
1: dell-wifi: Wireless LAN
        Soft blocked: no
        Hard blocked: yes
$ dkms status
bash: dkms: command not found
$ iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 00:1b:RE:DA:CT:ED
                type managed
$ iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), NO-IR
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
        (5170 - 5250 @ 80), (N/A, 20), (N/A), NO-IR
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, NO-IR
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
        (5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
$ lsmod | grep 88x
88x2bu               3047424  0
cfg80211              565248  4 iwl3945,iwlegacy,mac80211,88x2bu
$ grep -B1 -A4 idVendor=2357 /var/log/messages | cut -d' ' -f5-
usb 2-1: new high-speed USB device number 5 using ehci-pci
usb 2-1: New USB device found, idVendor=2357, idProduct=0126
usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 2-1: Product: 802.11n NIC
usb 2-1: Manufacturer: Realtek
usb 2-1: SerialNumber: 60A4REDACTED
$ lsusb | grep 2357:0126
Bus 003 Device 003: ID 2357:0126 TP-Link TL-WN8200N [Realtek RTL8812BU]
$ lspci | grep Net
09:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM5755M Gigabit Ethernet PCI Express (rev 02)
0c:00.0 Network controller: Intel Corporation PRO/Wireless 3945ABG [Golan] Network Connection (rev 02)
$ free
              total        used        free      shared  buff/cache   available
Mem:        8165488     2886024      765580      289700     4513884     4618236
Swap:       5001212      530124     4471088
$ grep ^model /proc/cpuinfo 
model           : 15
model name      : Intel(R) Core(TM)2 Duo CPU     T7100  @ 1.80GHz
model           : 15
model name      : Intel(R) Core(TM)2 Duo CPU     T7100  @ 1.80GHz
```

("Upgrade your kernel" isn't as easy as it sounds on this 14-year-old device; many times the graphics and/or network devices are unsupported in newer kernels. I do have dual root partitions, but sometimes even Grub gets messed up, so I'm not going to undertake an upgrade until I know I can take a week off work if necessary.)

morrownr, 2022-09-23:
Hi @kurahaupo 

> I need to make this work with an older 4.4 kernel, and I'm up for patching this rtl88x2bu driver if necessary

Alright then. After you have patched and tested, please return the favor by providing a pull request to update this repo.

Hint: In the directory where you cloned this driver, find and read the file called `supported-device-IDs`. I suspect you can take it from here.

Remember to run:

$ sudo ./remove-driver.sh

edit the appropriate file

$ sudo ./install-driver.sh

Remember to find a guide about how to prepare a pull request and then submit it.

Nick

kurahaupo, 2022-09-23:
okey-dokey!

Do you mind if I do a bit of tidying up as well?

kurahaupo, 2022-09-25:
Progress - it's now responding to USB probes.

However it now says that `rtw_usb_primary_adapter_init` fails; might that be because I already have a built-in (primary) WiFi card?

```
2022-09-25T16:45:07.815069+10:00 treehug kernel: [4604573.876205] usb 2-1: new high-speed USB device number 14 using ehci-pci
2022-09-25T16:45:07.947099+10:00 treehug kernel: [4604574.009306] usb 2-1: New USB device found, idVendor=2357, idProduct=0126
2022-09-25T16:45:07.947155+10:00 treehug kernel: [4604574.009319] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
2022-09-25T16:45:07.947165+10:00 treehug kernel: [4604574.009328] usb 2-1: Product: 802.11n NIC
2022-09-25T16:45:07.947170+10:00 treehug kernel: [4604574.009336] usb 2-1: Manufacturer: Realtek
2022-09-25T16:45:07.947175+10:00 treehug kernel: [4604574.009343] usb 2-1: SerialNumber: 60A4REDACTED
2022-09-25T16:45:07.947180+10:00 treehug kernel: [4604574.010702] RTW: 
2022-09-25T16:45:07.947185+10:00 treehug kernel: [4604574.010702] usb_endpoint_descriptor(0):
2022-09-25T16:45:07.947190+10:00 treehug kernel: [4604574.010711] RTW: bLength=7
2022-09-25T16:45:07.947195+10:00 treehug kernel: [4604574.010716] RTW: bDescriptorType=5
2022-09-25T16:45:07.947200+10:00 treehug kernel: [4604574.010722] RTW: bEndpointAddress=84
2022-09-25T16:45:07.947205+10:00 treehug kernel: [4604574.010727] RTW: wMaxPacketSize=512
2022-09-25T16:45:07.947210+10:00 treehug kernel: [4604574.010731] RTW: bInterval=0
2022-09-25T16:45:07.947215+10:00 treehug kernel: [4604574.010736] RTW: RT_usb_endpoint_is_bulk_in = 4
2022-09-25T16:45:07.947220+10:00 treehug kernel: [4604574.010741] RTW: 
2022-09-25T16:45:07.947226+10:00 treehug kernel: [4604574.010741] usb_endpoint_descriptor(1):
2022-09-25T16:45:07.947231+10:00 treehug kernel: [4604574.010747] RTW: bLength=7
2022-09-25T16:45:07.947236+10:00 treehug kernel: [4604574.010752] RTW: bDescriptorType=5
2022-09-25T16:45:07.947241+10:00 treehug kernel: [4604574.010757] RTW: bEndpointAddress=5
2022-09-25T16:45:07.947246+10:00 treehug kernel: [4604574.010761] RTW: wMaxPacketSize=512
2022-09-25T16:45:07.947251+10:00 treehug kernel: [4604574.010766] RTW: bInterval=0
2022-09-25T16:45:07.947256+10:00 treehug kernel: [4604574.010771] RTW: RT_usb_endpoint_is_bulk_out = 5
2022-09-25T16:45:07.947261+10:00 treehug kernel: [4604574.010776] RTW: 
2022-09-25T16:45:07.947266+10:00 treehug kernel: [4604574.010776] usb_endpoint_descriptor(2):
2022-09-25T16:45:07.947271+10:00 treehug kernel: [4604574.010782] RTW: bLength=7
2022-09-25T16:45:07.947276+10:00 treehug kernel: [4604574.010787] RTW: bDescriptorType=5
2022-09-25T16:45:07.947281+10:00 treehug kernel: [4604574.010792] RTW: bEndpointAddress=6
2022-09-25T16:45:07.947286+10:00 treehug kernel: [4604574.010796] RTW: wMaxPacketSize=512
2022-09-25T16:45:07.947291+10:00 treehug kernel: [4604574.010801] RTW: bInterval=0
2022-09-25T16:45:07.947296+10:00 treehug kernel: [4604574.010805] RTW: RT_usb_endpoint_is_bulk_out = 6
2022-09-25T16:45:07.947301+10:00 treehug kernel: [4604574.010810] RTW: 
2022-09-25T16:45:07.947306+10:00 treehug kernel: [4604574.010810] usb_endpoint_descriptor(3):
2022-09-25T16:45:07.947311+10:00 treehug kernel: [4604574.010817] RTW: bLength=7
2022-09-25T16:45:07.947316+10:00 treehug kernel: [4604574.010821] RTW: bDescriptorType=5
2022-09-25T16:45:07.947321+10:00 treehug kernel: [4604574.010826] RTW: bEndpointAddress=87
2022-09-25T16:45:07.947326+10:00 treehug kernel: [4604574.010831] RTW: wMaxPacketSize=64
2022-09-25T16:45:07.947331+10:00 treehug kernel: [4604574.010835] RTW: bInterval=3
2022-09-25T16:45:07.947336+10:00 treehug kernel: [4604574.010841] RTW: RT_usb_endpoint_is_int_in = 7, Interval = 3
2022-09-25T16:45:07.947341+10:00 treehug kernel: [4604574.010845] RTW: 
2022-09-25T16:45:07.947346+10:00 treehug kernel: [4604574.010845] usb_endpoint_descriptor(4):
2022-09-25T16:45:07.947351+10:00 treehug kernel: [4604574.010852] RTW: bLength=7
2022-09-25T16:45:07.947356+10:00 treehug kernel: [4604574.010856] RTW: bDescriptorType=5
2022-09-25T16:45:07.947361+10:00 treehug kernel: [4604574.010861] RTW: bEndpointAddress=8
2022-09-25T16:45:07.947366+10:00 treehug kernel: [4604574.010866] RTW: wMaxPacketSize=512
2022-09-25T16:45:07.947371+10:00 treehug kernel: [4604574.010870] RTW: bInterval=0
2022-09-25T16:45:07.947376+10:00 treehug kernel: [4604574.010875] RTW: RT_usb_endpoint_is_bulk_out = 8
2022-09-25T16:45:07.947381+10:00 treehug kernel: [4604574.010880] RTW: nr_endpoint=5, in_num=2, out_num=3
2022-09-25T16:45:07.947386+10:00 treehug kernel: [4604574.010880] 
2022-09-25T16:45:07.947391+10:00 treehug kernel: [4604574.010887] RTW: USB_SPEED_HIGH
2022-09-25T16:45:07.947396+10:00 treehug kernel: [4604574.010892] RTW: CHIP TYPE: RTL8822B
2022-09-25T16:45:07.947401+10:00 treehug kernel: [4604574.011027] RTW: [HALMAC]55772M
2022-09-25T16:45:07.947406+10:00 treehug kernel: [4604574.011027] HALMAC_MAJOR_VER = 1
2022-09-25T16:45:07.947411+10:00 treehug kernel: [4604574.011027] HALMAC_PROTOTYPE_VER = 6
2022-09-25T16:45:07.947416+10:00 treehug kernel: [4604574.011027] HALMAC_MINOR_VER = 6
2022-09-25T16:45:07.947421+10:00 treehug kernel: [4604574.011027] HALMAC_PATCH_VER = 23
2022-09-25T16:45:07.951063+10:00 treehug kernel: [4604574.012914] RTW: ERROR [HALMAC][ERR]Chip id is undefined
2022-09-25T16:45:07.951101+10:00 treehug kernel: [4604574.012924] RTW: ERROR rtw_halmac_init_adapter: halmac_init_adapter fail!(status=54)
[... cut ...]
2022-09-25T16:45:07.951314+10:00 treehug kernel: [4604574.013168] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_inprogress ###
2022-09-25T16:45:07.951319+10:00 treehug kernel: [4604574.013174] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_mac_register ###
2022-09-25T16:45:07.951323+10:00 treehug kernel: [4604574.013179] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_phy ###
2022-09-25T16:45:07.951328+10:00 treehug kernel: [4604574.013208] RTW: rtw_usb_primary_adapter_init Failed!
```

kurahaupo, 2022-09-25:
Ok, this just gets weirder. I threw in a debug "printf" and got out:

```
2022-09-25T17:09:24.466560+10:00 treehug kernel: [4606030.527003] RTW: WARN [HALMAC][INFO] chip_id=0x07, chip_ver=0x01, halmac_chip_id=0x7f
2022-09-25T17:09:24.466563+10:00 treehug kernel: [4606030.527007] RTW: ERROR [HALMAC][ERR]Chip id 0x07 is undefined
```
and it turns out that Chip id 7 is `CHIP_ID_HW_DEF_8192E`, which does not appear to have any code in the Halmac section.

So now I'm wondering if it's neither RTL8812 nor RTL8822 based at all?
(Switches the `rtw_usb_id_tbl` entry from RTL88x2BU to RTL_8192E ... and nope, that's back to completely unresponsive.)

(As you can probably surmise from me babbling away with updates, I haven't done kernel patching in a very long time, like before USB was a thing. Feel free to ignore these notes.)

morrownr, 2022-09-25:
Martin,

After rereading some of your comments, I did some research. The device ID, 2357:0126, appears to indicate your chipset is RTL8192EU. I think the driver for that chipset is in modern Linux kernels so you don't even need to find or install a driver.

I have hardware as old as yours still running. What you could do is burn a copy of Xubuntu 22.04 LTS to a usb flash driver and boot the system. You can then see how everything works and make a decision whether to install it or not. You might be pleasantly surprised. If you have problems, let me know and I will see if I can help.

FYI: The Main Menu of this site is:

https://github.com/morrownr/USB-WiFi

There is some good information here.

Regards,

Nick

kurahaupo, 2022-09-26:
Thankyou for that. I will look closer at Ubuntu 2022.04 LTS. (I also discovered that running "depmod -a" before rebooting solves a bunch of weird stuff _after_ booting; in particular the `iwl3945` driver is tagged "legacy" and so not loaded by default.)

Back to this module, adding a (vacuous) entry for chip ID 7 resulted in a kernel panic when I plugged in the device, so unless I made a stupid error somewhere else, that points to this being the wrong development direction.

The odd thing about all of this is that I arrived at your Github repo from TP-Link's community pages, as there seemed to be a number of indications that the 88x2 drivers were appropriate for a number of chipset families.

Thanks again for all your help. I've left the proposed patch available as it has a number of structural improvements that you might like to incorporate, but don't bother with the final commit 643b6e6 as it seems to make the system unstable/broken.

closed 2022-09-26 by kurahaupo

morrownr, 2022-09-26:
> I will look closer at Ubuntu 2022.04 LTS.

I would not do that. The desktop environment on Ubuntu is too heavy for a 14 year old system. On the other hand, Xubuntu uses a lighter desktop environment and could work well.

> The odd thing about all of this is that I arrived at your Github repo from TP-Link's community pages, as there seemed to be a number of indications that the 88x2 drivers were appropriate for a number of chipset families.

That is bad information. There is a lot of bad information on the net. The 88x2 driver is good for adapters that have 8812bu and 8822bu chipsets. The 8822bu is the 8812bu plus bluetooth. Some may have come up with the idea after looking at the source but Realtek keeps information about several chipsets in some files but don't let that fool you. 

> I've left the proposed patch available as it has a number of structural improvements

Thank you. I see some good ideas. I have already patched the duplicate device ID issue. I will hold your patches to go over them. This repo gets over 250 hits per day so I have to be very careful not to break things. I may close your PR but will study it in the background.

Let me finish by reiterating that I have a Main Menu here that provides a lot of info. Many Linux users do not realize that the Realtek out-of-kernel drivers are not Linux standards compliant which causes problems. Most Linux users would find adapters that use in-kernal drivers to be the better choice. Currently we have new adapters that use the new mt7921au chipset coming on the market. That chipset is WiFi 6e capable and the driver (mt7921u) is already in recent kernels.

https://github.com/morrownr/USB-WiFi

Cheers

kurahaupo, 2022-09-26:
One idea I was wondering about was to swap around the order of the enums so
that "invalid" would be 0. Then lookup tables can be sparsely populated
with the correct result for unspecified elements.

Currently they all appear to use max-signed-value for whatever size the
enum fits into (0x7f, 0x7fff, or 0x7fffffff), even though the enum may in
fact be unsigned; but as far as I can see no code actually makes any use of
that fact.

In my patches I use the GCC array initialiser extension { [0 ... MAX] =
UNDEF, [EnumValue] = Result, } which isn't terrible because the kernel
already uses lots of GCC extensions, but isn't ideal because it means
guessing the appropriate max value.