morrownr/88x2bu-20210702 #84: Limited Connectivity on 5Ghz

rmjantz, 2022-10-01:
I am using the driver with an EDUP 1300AC dongle and OpenSuse Tumbleweed.  2.4Ghz works great, but 5Ghz has issues.  The most common symptom is that NetworkManager says 'limited connectivity.'  Sometimes it says 'full connectivity,' but then it takes forever to render a web page - rendering the web page may fail and it will go back to limited connectivity.  Once, after resetting the router, 5Ghz worked great for a couple of days, getting the kind of speed I expect, but then it went back to limited connectivity.  I reset the router again, thinking it would bring the 5Ghz back, but but after running one successful speed test, it went back to limited connectivity.  This is a dual boot system and 5Ghz seems to have no issues on Windows 10.

I am happy to do more diagnosis myself if I can get some tips on things I should try.

morrownr, 2022-10-01:
Hi @rmjantz 

I'll try to help.

The first thing to do is go to top of the issues page and supply the information requested in `Problem Reporting Guidelines."

It would be good for me to know what distros and version you are using.

Often when I problems in 5 GHz while 2.4 is working well, I look first at two things:

- Is the country code set properly:

$ iw reg get

Please post the results of the above and also tell me what country you are in.

- There is also the issue of signal strength because 5GHz just does not have the range that 2.4 does. If you install `wavemon`, we could get some data that might help us.

$ sudo apt install wavemon
$ wavemon

With wavemon, check both bands and tell me what the signal strength is.

Do both bands use the same name? If so, try changing the name of the 5 GHz band let's see if that helps.

Also, give me the results of:

$ lsusb -t

That is enough to start.

Cheers

rmjantz, 2022-10-02:
Thanks for the quick reply.  Here is the requested info:

Linux distro:  OpenSuse Tumbleweed version 5.19.10-1-default

Name and model of computer:  Lenovo E420 laptop

country:  United States

lsusb -t output:
rjantz@localhost:~> lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 4: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 4: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 4: Dev 3, If 2, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 4: Dev 3, If 3, Class=Application Specific Interface, Driver=, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M

wavemon signal level 2.4Ghz:  signal level: -32 dBm (0.63 uW)
wavemon signal level 5Ghz:  signal level: -28 dBm (1.58 uW)

Do both bands use the same name:  no

Info from reporting guidelines
rjantz@localhost:~> sudo uname -mr
5.19.10-1-default x86_64
rjantz@localhost:~> sudo  mokutil --sb state
This system doesn't support Secure Boot
rjantz@localhost:~> sudo lsusb
Bus 002 Device 003: ID 0a5c:217f Broadcom Corp. BCM2045B (BDC-2.1)
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 04f2:b257 Chicony Electronics Co., Ltd Lenovo Integrated Camera
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rjantz@localhost:~> sudo rfkill list all
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rjantz@localhost:~> sudo dkms status
rtl88x2bu, 5.13.1, 5.19.10-1-default, x86_64: installed
rjantz@localhost:~> sudo iw dev
phy#0
        Interface wlp0s26u1u1
                ifindex 3
                wdev 0x1
                addr e8:4e:06:9f:2d:aa
                ssid JaRay2
                type managed
                channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 15.00 dBm
rjantz@localhost:~> sudo iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

rmjantz, 2022-10-02:
For some reason, it used strikethrough on part of the text in my last post.  I am reposting that part.

rjantz@localhost:~> sudo uname -mr
5.19.10-1-default x86_64
rjantz@localhost:~> sudo lsusb
Bus 002 Device 003: ID 0a5c:217f Broadcom Corp. BCM2045B (BDC-2.1)
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 04f2:b257 Chicony Electronics Co., Ltd Lenovo Integrated Camera
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rjantz@localhost:~> sudo dkms status
rtl88x2bu, 5.13.1, 5.19.10-1-default, x86_64: installed

rmjantz, 2022-10-02:
Hmm... more strikeout - guess I'll leave well enough alone

morrownr, 2022-10-03:
Don't worry about the strikeout. 

> country 00: DFS-UNSET

This is a concern. Try this:

$ iw reg set US
$ iw reg get

That should set your country code and that should allow you to use 5 GHz.

Also,

>  Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M

The adapter is running in USB2 mode. That can be caused plugging it into a USB2 port or by not setting the option to use USB3. You may not need the extra speed. Using the `./edit-options.sh` script will help you set USB3. You will need to read the docs in the file that comes up and set the appropriate options in the `options` line.

Let me know how it goes.

rmjantz, 2022-10-03:
Below is the result.  Still limited connectivity, I'm afraid.

rjantz@localhost:~> sudo iw reg set US
[sudo] password for root: 
rjantz@localhost:~> sudo iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

rjantz@localhost:~> sudo iwconfig
lo        no wireless extensions.

enp2s0    no wireless extensions.

wlp0s26u1u1  IEEE 802.11AC  ESSID:"JaRay2"  Nickname:"WIFI@RTL88X2BU"
          Mode:Managed  Frequency:5.745 GHz  Access Point: CC:40:D0:7D:AB:63   
          Bit Rate:867 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=59/100  Signal level=65/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

rjantz@localhost:~> nmcli n connectivity
limited

morrownr, 2022-10-03:
> ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

I saw this earlier so yes, you do have a usb wifi adapter based on the 8812bu chipset so that this the right driver.

With that said, something is wrong.

> wlp0s26u1u1 IEEE 802.11AC ESSID:"JaRay2" Nickname:"WIFI@RTL88X2BU"

The ` wlp0s26u1u1` interface name tells me this is an internal PCI card. However, it is trying to use the driver here which should not happen as this driver is for USB only. Did you change any settings other than what the installation steps called for?

 > Link Quality=59/100 Signal level=65/100 Noise level=0/100

Those numbers are not good but that is beside the point until we figure out what is going on.

Run and post the results of:

$ lspci

rmjantz, 2022-10-03:
Installation was by the book (install-driver.sh) and it went smooth.  I neglected to do the OS upgrade ahead of the install, so I did it afterward.  I was previously using a TP-Link USB adapter and it used the same device name.  Below is lspci output.

rjantz@localhost:~> lspci
00:00.0 Host bridge: Intel Corporation 2nd Generation Core Processor Family DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
00:16.0 Communication controller: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 (rev b4)
00:1c.1 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 2 (rev b4)
00:1c.2 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 3 (rev b4)
00:1c.7 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 8 (rev b4)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation HM65 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 6 Series/C200 Series Chipset Family 6 port Mobile SATA AHCI Controller (rev 04)
00:1f.3 SMBus: Intel Corporation 6 Series/C200 Series Chipset Family SMBus Controller (rev 04)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
03:00.0 System peripheral: Ricoh Co Ltd MMC/SD Host Controller (rev 07)

morrownr, 2022-10-03:
Well, okay. I fishing at this point.

$ dkms status

Let's see what the above says.

rmjantz, 2022-10-03:
See below.

rjantz@localhost:~> sudo dkms status
[sudo] password for root: 
rtl88x2bu, 5.13.1, 5.19.10-1-default, x86_64: installed

morrownr, 2022-10-04:
I just took the time to scan the thread and rethink this issue.

> Link Quality=59/100 Signal level=65/100

I'm pretty sure this is the core problem. A signal level of 65 is right at the end of usability. Levels < 50 are good, > 50 and < 60 are not to bad but >= 65 is bad. Why it could be different than you see in Windows is unknown. What are your options:

- Run wavemon and move around. See how the readings change. It might turn out that a little move would help.
- You can try turning up the txpower. Many adapters have fixed settings in this day in time so you would just have to try.
- Change port. 
- Reposition your AP.
- All things equal, channel 36 should give a slightly better signal than 149 due to the lower freq.

rmjantz, 2022-10-05:
Here's signal level per wavemon:
2.4Ghz -33dbM
5Ghz -38 dbM
Seems OK?

I'm already pretty close to the router, but I moved right next to it and I think signal level for 5Ghz was in the high 20's

I rebooted the router again and 5Ghz came back (using it right now).  So 2 out of 3 tries of a reboot helped.  Is there a clue in there?

FYI, Windows (and even my phone) seem to mostly work well, but every now and then, even they get flaky and can't connect.

I am looking into upping the tx power.

morrownr, 2022-10-05:
> Seems OK?

Yes

> I am looking into upping the tx power.

I would not bother. Your posted info indicates the signal is not the issue.

> I rebooted the router again and 5Ghz came back (using it right now). So 2 out of 3 tries of a reboot helped. Is there a clue in there?

Probably so. Can you check to see if there is a firmware upgrade available for your router? If so, maybe you should install it.

It the problem is still there. it could be an issue with one or more settings in the router. I have a short section in the README about router settings. You can take a look at that.

rmjantz, 2022-10-05:
The router is a Netgear Nighthawk R7000.  It has firmware version 1.0.11.126.  It has auto update enabled, but the Netgear site says the latest is 1.0.11.136.   When I tell the router to check for updates it says none found, but apparently others have seen this behavior too, even though the website says there is later firmware.  I think I should try this.  Q:  Is it recommended to do this over a wired connection?

morrownr, 2022-10-05:
> Q: Is it recommended to do this over a wired connection?

Yes, wired would be best. It sounds like the auto-update is not working as it should be.

After you are updated, we could discuss settings.

rmjantz, 2022-10-06:
Changed the firmware - same symptoms, 5Ghz works sometimes, most of the time not.  Do you think settings could make a difference?

morrownr, 2022-10-06:
>  Do you think settings could make a difference?

Yes. The problem is that I have no idea how your router settings pages are set up so maybe just posting a few at a time for discussion might work.

rmjantz, 2022-10-07:
Just letting you know I'll be out of your hair for a week since I'm going on a family vacation.  FYI, 5Ghz is working good at the moment and I haven't changed anything.  Thanks for your help.

morrownr, 2022-10-09:
Let me know if the problems return. Enjoy the vacation.

rmjantz, 2022-10-20:
Hi,
Back from vacation.   Here's where I think we are:
- Router reboot pretty consistently makes 5Ghz work again for a couple of days
- After a couple of days I get get limited connectivity again
- My phone, tablet, and firestick seem to work OK even when I'm getting limited connectivity through my dongle
- I tried to trace things with wireshark and when it's not working I see a lot of TCP resets, retransmissions , and destination unreachable (I think - I'm not a wireshark expert).
- I found a cheap dongle with MT7610U chipset which appears to be supported by tumbleweed
- I will give that a try when it arrives and let you know how it works - no need to do any more till then

morrownr, 2022-10-21:
Copy all.

rmjantz, 2022-10-30:
Hi,
We're probably at a place where we can close this - here's where things stand:  While I was waiting on the mt7610U-based adapter (and still getting limited connectivity), I figured I'd try another router reboot.  I rebooted it and the 5Ghz connectivity came back.  I thought it would only last a couple of days (which has been my past experience), but it's been working great ever since (about a week).  The only other things I've done are a remove/reinstall of the driver and and an upgrade to linux 6.0.

I received the mt7610U--based dongle and plugged it in.  It worked fine with no driver install needed.  I thought 5Ghz speeds were a little lower (150 Mbps vs 200 Mbps with the RTL8812bu), but then I tried the RTL8812bu dongle and it was also getting lower speeds.  And I just tested the mt7610U and its speeds were close to 200 Mbps.  What I think this tells me is (a) 5Ghz performance and reliability is susceptible to a lot of random factors that are hard to diagnose and (b) wide variation in 5Ghz speeds is normal.  Now that I have an adapter with an in-kernel driver, I will make that one the one I use and the other a backup.  I might solicit your advice on one more thing:  the  LED on the 7610 adapter isn't working, even though it works on Windows.  Any idea why that is?  I notice you address a similar issue for the 7612U driver.  Could that apply to the 7610U as well?  If I don't get the LED working, no big deal.  Thanks for all your help

morrownr, 2022-10-31:
> an upgrade to linux 6.0.

What distro and version are you running. Something that I am aware of since I test so much is that upgrading a system to a very recent kernel where the rest of the distro remains as it was can result in strange happenings since so much of the supporting utilities and other parts of the distro are tested extensively with the kernel that ships with the distro but not with the latest mainline.

> I received the mt7610U--based dongle and plugged it in. It worked fine with no driver install needed.

You will probably find this adapter to be handy even if you don't need it right now. I've keep adapters that use in-kernel drivers in my toolkit for years.

> And I just tested the mt7610U and its speeds were close to 200 Mbps.

What it tells me is that you are likely on a 5 GHz channel that is fairly heavily congested. This is a sharing technology so other packets going through air on the same channel share the same bandwidth that you are using.

> 5Ghz performance and reliability is susceptible to a lot of random factors that are hard to diagnose

Well, I'm going to disagree a little. It can take a certain degree of technical experience to maximize performance but it can be done.

> wide variation in 5Ghz speeds is normal

I agree with this statement in general but I will add that there are reasons for the variations and there are generally solutions to increase speeds and reduce problems. Go to:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for this site. You are welcome to look around but for now, go to Issues at the Main Menu repo and look at 139 (Speed Test with Specific USB WiFi Adapters).

Take a look at the speeds. Wow... high speeds that are totally stable. Tell me how I did that?

Nick

rmjantz, 2022-11-10:
I am closing this.  Problem seems to have gone away for unknown reasons.  In any case it does not seem to be a driver issue.

closed 2022-11-10 by rmjantz

morrownr, 2022-11-11:
> Mine are a lot more erratic, but seem seem to get in the neighborhood of 200 Mbps around half the time. I don't see a lot of evidence of competing networks using Wifi Analyzer on my phone. Would that reveal congestion?

Somehow I forgot to answer this question so here goes.

A WiFi Analyzer app can give an idea but not the exact answer. All it is telling you is how many AP's it is seeing. It is not telling your how much data is being pushed between systems on the channel you are on and that is what you need to know to determine congestion.

The reason you saw the vert fast, very consistent speeds in the test I posted is because I set the channel in my router to a DFS channel where there is no other AP. The air is clean. There is no congestion. The results in the test are what the adapters can do with my router without congestion. The same Alfa ACM you saw getting around 400 Mb/s will show around 200 Mb/s if I change the channel to 36 (a regular heavily congested channel.) You can judge congestion by seeing what you get vs what the adapter can get with no congestion.

Cheers,

Nick