morrownr/88x2bu-20210702 #86: the driver for linksys WUSB6300V2 isn't works

Azaretdodo, 2022-10-03:
hello,

the driver for linksys WUSB6300V2 isn't works,

thanks you in advance to bring it,

regards.

Azaretdodo.

morrownr, 2022-10-03:
Hi @Azaretdodo 

There is an Issue called `Problem Reporting Guidelines - Please Read` that is pinned at the top of the main Issues page. Please read and follow the instructions. We will need more information to help you solve your problem.

Regards

Azaretdodo, 2022-10-04:
hello,

i paste here the output of the line of command asked by your guide : 
```

sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.19.0-15-generic x86_64
EFI variables are not supported on this system
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 04ca:703c Lite-On Technology Corp. Integrated Camera
Bus 002 Device 004: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 002 Device 003: ID 17ef:608d Lenovo Optical Mouse
Bus 002 Device 006: ID 13b1:0045 Linksys WUSB6300 V2
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: tpacpi_bluetooth_sw: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8812au/5.13.6, 5.19.0-15-generic, x86_64: installed
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr 94:65:9c:27:37:ac
		type P2P-device
		txpower 0.00 dBm
	Interface wlp3s0
		ifindex 3
		wdev 0x1
		addr 94:65:9c:27:37:ab
		ssid SFR_80A0_5GHZ
		type managed
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0 (self-managed)
country 00: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2472 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(2457 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN
	(5170 - 5190 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5190 - 5210 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5210 - 5230 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5230 - 5250 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5250 - 5270 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5270 - 5290 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5290 - 5310 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5310 - 5330 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5490 - 5510 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5510 - 5530 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5530 - 5550 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5550 - 5570 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5570 - 5590 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5590 - 5610 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5610 - 5630 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5630 - 5650 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5650 - 5670 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5670 - 5690 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5690 - 5710 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5710 - 5730 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5815 - 5835 @ 20), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN

```
thanks you in advance to repair my issue,

regards.

Azaretdodo.

morrownr, 2022-10-04:
Good day @Azaretdodo 

Okay, now I can see some problems. 

> rtl8812au/5.13.6, 5.19.0-15-generic, x86_64: installed

This line tells me you have installed the driver for the 8812au chipset. That is not the driver in this repo.

> ID 13b1:0045 Linksys WUSB6300 V2

This line tells me the chipset in your adapter is the 8812bu. The driver in this repo is the right one.

> country 00: DFS-UNSET

This line tells me your country code is not set which may lead to 5 GHz problems once the right driver is installed. I can help with this once the correct driver is installed if you want.

You need to remove the 8812au driver: (go to the driver directory)

$ sudo ./remove-driver.sh

Delete the 8812au driver directory and then do the installation steps for the 88x2bu driver (the one in this directory).

Nick

Azaretdodo, 2022-10-04:
Dear nick,

Can you bring the readme for this driver (88x2bu driver) because I have keep up the good readme and the good directory unfortunately this isn't good thus thanks you in advance to bring a real help,

Regards.

Azatetdodo.

Dorian Rosse.

Azaretdodo, 2022-10-04:
> Hi Dorian,
>
> I'm not sure if I understand the question.
>
> What has happened to you is that you have installed the wrong driver and it will need to be removed.
>
> The README for the correct driver is located on the main page:
>
> https://github.com/morrownr/88x2bu-20210702
>
> Scroll down.
>
> I realize the README is long but I am not sure if there is a solution to that for Realtek drivers.

does this fork github is the good ? : 88x2b

i was think there are only one fork but i have see this one,

thanks you in advance for your answer,

regards.

Azaretdodo

morrownr, 2022-10-04:
> does this fork github is the good ? : 88x2b

88x2bu is a placeholder that points to this repo.

Azaretdodo, 2022-10-04:
i have instal it then i have reboot for finaly put the wireless password finaly i close the ticket this is good for my problem,

regards.

Azaretdodo.

closed 2022-10-04 by Azaretdodo