morrownr/88x2bu-20210702 #87: troubles to set country code

pmirabel, 2022-10-04:


morrownr, 2022-10-04:
Not much to go on here. What have you tried and what distro are you using?

pmirabel, 2022-10-05:
I am so sorry I was editing this issue and accidentally published it blank ! 
I edited my message ! Thx

morrownr, 2022-10-05:
Good day @pmirabel 

I see. Well, that happens. Let's see what we can do.

> 4.19.73-intel-pk-standard x86_64

Custom old kernel. We may want to ensure the proper regulatory db files are in place.

Can you install `git`? If so:

```
Updating the WiFi Regulatory db

Webpage:

 git://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git

Download:

 git clone git://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git

If you have kernel version 4.15 or newer, the regulatory database is loaded using the same mechanism as the firmware files. The files can be found at

https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/

You'll need two files, regulatory.db and regulatory.db.p7s. Place them into /lib/firmware/ directory.

$ sudo cp regulatory.db /lib/firmware/
$ sudo cp regulatory.db.p7s /lib/firmware/
$ sudo reboot

```

Once that is complete:

$ iw reg set US
change US to your country code

$ iw reg get
to see if it worked

------

> So far, I wasn't able to get 88x2bu.conf shipped in my OS at the right spot /etc/modprobe.d

From the driver directory:

$ sudo cp 88x2bu.conf /etc/modprobe.d

To edit:

$ sudo nano /etc/modprobe.d/88x2bu.conf

reboot for the file to take effect.

-----

Let me know if you are making progress.

Nick

pmirabel, 2022-10-06:
Thanks a lot, your message made me realise that I was missing `wireless-regdb` package...I added it to my OS and that work now, at least for FR / DE countries.
I think I still have an issue with CA : 
```
iw reg set CA
iw reg get 
    global
    country 98: DFS-UNSET
```
Indeed, I thought it would be : 
```
global
country CA: DFS-FCC
```

NB : Still digging with modprobe issue, will keep you updated

morrownr, 2022-10-06:
Sometimes it helps if you add the `rtw_country_code` driver option (module parameter).

rtw_country_code=CA

Were you able to get the file `88x2bu.conf` to the right place?

morrownr, 2022-10-09:
@pmirabel 

Were you able to make any progress?

closed 2022-10-19 by morrownr

pmirabel, 2023-04-03:
Just to let you know, I fixed my issue by : 
- upgrade my distribution kernel to 5.4
- modified my bitbake recipe to look like : 
```
inherit module externalsrc

KO_NAME = "88x2bu"
KO_DIR = "${base_libdir}/modules/${KERNEL_VERSION}/kernel/drivers/net/wireless/${PN}"

# Kernel module packages MUST begin with 'kernel-module-', otherwise
# multilib image generation can fail.
#
# The following line is only necessary if the recipe name does not begin
# with kernel-module-.
PKG_${PN} = "kernel-module-${KO_NAME}"
# FILES_${PN} += "/etc/modprobe.d/*"
EXTERNALSRC = "${THISDIR}/files/tplink-v3"
EXTERNALSRC_BUILD = "${EXTERNALSRC}"

EXTRA_OEMAKE += "KSRC=${STAGING_KERNEL_DIR}"

do_install() {
    install -d ${D}${KO_DIR}
    install -p -m 644 ${KO_NAME}.ko ${D}${KO_DIR}
    mkdir -p ${D}/etc/modprobe.d
    install -m 644 ${EXTERNALSRC}/${KO_NAME}.conf ${D}/etc/modprobe.d
}
RPROVIDES_${PN} += "kernel-module-${KO_NAME}"
KERNEL_MODULE_AUTOLOAD += "${KO_NAME}"
MACHINE_EXTRA_RRECOMMENDS = " kernel-modules"
# KERNEL_MODULE_PROBECONF += "${KO_NAME}"
# module_conf_${KO_NAME} = "options 88x2bu rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_switch_usb_mode=0 rtw_country_code=FR"
```

Thanks again for your help!