morrownr/88x2bu-20210702 #90: Builds but does not function on RHEL8.6 / Rocky Linux 8.6

tmcqueen-materials, 2022-10-23:
Thank you very much for maintaining this driver. This may be an impossibility, but I was hoping that this driver would function on RHEL 8.x and clones. While it seems to build, install, and load no problem on 8.6, the driver does not appear to transmit any packets, instead all packets are dropped (receiving packets seems to work no problem). This is operating in IBSS mode.

Things I have tried:
1. Multiple different USB ports (both USB 2 and USB 3), across two different computers running RHEL 8.6. No variation work.
2. Changing to a functional wifi adapter, but keeping the same wpa_supplicant, etc, configuration. This works.
3. Use an rtl8188-based device with a corresponding driver. This works, when using the older driver ( https://github.com/lwfinger/rtl8188eu/tree/v4.1.8_9499 branch), with some version checks tweaked so it builds on RHEL 8.6.
4. The same rtl8188-based device above does not work when using the 5.x driver ( https://github.com/lwfinger/rtl8188eu ), with version checks tweaked so it builds on RHEL 8.6. I mention this because the failure mode is the same as with the 88x2bu driver here -- receives work fine, but all transmitted packets are dropped. 

The above suggest to me some kernel backport is screwing up frame transmissions. Unfortunately, I was unable to figure out where the problem comes from. Do you happen to have any idea what the culprit might be? Some logs below might be helpful.

Output of "report a problem" command:
```
4.18.0-372.26.1.el8_6.x86_64 x86_64
SecureBoot disabled
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 04b8:015c Seiko Epson Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0764:0501 Cyber Power System, Inc. CP1500 AVR UPS
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 002: ID 05ac:1500 Apple, Inc. SuperDrive [A1379]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
10: phy11: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl88x2bu/5.13.1, 4.18.0-372.26.1.el8_6.x86_64, x86_64: installed
phy#11
	Interface wlp4s0f3u2
		ifindex 17
		wdev 0xb00000001
		addr 34:60:f9:93:b9:98
		ssid SensorMesh11
		type IBSS
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 16.00 dBm
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

Relevant ifconfig interface:
```
wlp4s0f3u2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.11.254  netmask 255.255.255.0  broadcast 192.168.11.255
        ether 34:60:f9:93:b9:98  txqueuelen 1000  (Ethernet)
        RX packets 1  bytes 126 (126.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 38 overruns 0  carrier 0  collisions 0
```

dmesg with driver at log level info:
```
[1375600.080286] RTW: module init start
[1375600.080288] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[1375600.080289] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[1375600.080301] RTW: rtw_inetaddr_notifier_register
[1375600.080322] RTW: 
                 usb_endpoint_descriptor(0):
[... 332 lines cut ...]
[1375637.916112] RTW: ### Set STA_(0) info ###
[1375637.916839] RTW: ============ STA [b8:27:eb:95:9f:c6]  ===================
[1375637.916843] RTW: mac_id : 0
[1375637.916844] RTW: wireless_mode : 0x03
[1375637.916845] RTW: mimo_type : 0
[1375637.916847] RTW: static smps : N
[1375637.916848] RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
[1375637.916850] RTW: rate_id : 0
[1375637.916851] RTW: rssi : -1 (%), rssi_level : 0
[1375637.916854] RTW: is_support_sgi : N, is_vht_enable : N
[1375637.916855] RTW: disable_ra : N, disable_pt : N
[1375637.916857] RTW: is_noisy : N
[1375637.916858] RTW: txrx_state : 0
[1375637.916860] RTW: curr_tx_rate : CCK_1M (L)
[1375637.916862] RTW: curr_tx_bw : 20MHz
[1375637.916863] RTW: curr_retry_ratio : 0
[1375637.916864] RTW: ra_mask : 0x0000000000000fff

[1375637.916867] RTW: rtw_set_rts_bw connect_to_8812=0,enable=1
[1375639.606641] RTW: WARN Invalid hw_rate 0xff in hw_rate_to_m_rate
[1375639.607631] RTW: WARN Invalid hw_rate 0xff in hw_rate_to_m_rate
[1375664.184330] RTW: WARN Invalid hw_rate 0xff in hw_rate_to_m_rate
[1375664.185362] RTW: WARN Invalid hw_rate 0xff in hw_rate_to_m_rate
[1375722.121169] RTW: cfg80211_rtw_get_txpower(wlp4s0f3u2) total max: 1600 mbm
```

morrownr, 2022-10-23:
Hi @tmcqueen-materials 

> The above suggest to me some kernel backport is screwing up frame transmissions.

You are likely correct. I attempt to make this and the other Realtek drivers here usable on as many distros as possible. I keep a small lab with test systems to help in that regard. Some distros are hard to support and RHEL is one of them. RHEL adds several roadblocks that would take a lot of time to surmount and it is beyond the time I have available for this project, Signing up as a dev or paying for something I don't need just to get a copy is something that I don't need. What I do is allow those who do have the time and knowledge to provide PRs to add or maintain RHEL support. I then test to make sure they don't break other distros and if not, everyone is happy. This driver had support for RHEL 8.4 added at some point last year. 

What RHEL does with extensive back porting makes it extremely hard to with drivers such as the one here. You might make headway by using a tool such as `meld` to view the differences between a Linux mainline kernel source and the RHEL 8.6 source and then compare the changes to what has already be done in this source. 

Another possible option: The Linux mainline source for kernel 4.19 contains the in-kernel driver mt7612u . You might patch it into the 4.18 source and see what happens. I can show you where the firmware is kept on linux.com.

Another possible option: Move to RHEL9. It appears to use kernel 5.14 and should include in-kernel drivers for several usb wifi adapter that are dual band.

You can see the various adapters that use in-kernel drivers by selecting menu item 2 at the Main Menu:

https://github.com/morrownr/USB-WiFi

@morrownr

tmcqueen-materials, 2022-10-23:
@morrownr Thanks! That is what I suspected. If/When I have time, I'll see if I can figure it out (forcing rtw_linked_check to return true avoids the drops, but still doesn't seem to allow proper bidirectional packet delivery, so I presume its check is more of a symptom than the underlying cause).

tmcqueen-materials, 2022-10-23:
@morrownr : actually, I hadn't tested properly. Simply patching rtw_linked_check to always return true makes it function. Given the check is functionally the same in both the rtl8188eu driver that works and the rtl8188eu (and 88x2bu driver) that don't, and that this appears mostly to be accounting inside the driver, no idea what in the RHEL 8.6 kernel screws that up. But for future folks, modify `core/rtw_mlme.c` to add these four lines at the top of the rtl_linked_check function:
```
#if defined(RHEL8)
        // Some part of this check is broken on RHEL 8.6. Override
        return _TRUE;
#endif
```

tmcqueen-materials, 2022-10-23:
@morrownr As a final comment, here is a less invasive patch that seems to fix the issues. Something about station accounting is all messed up on RHEL 8.6 -- the drivers that work are not checking for a link before sending/receiving, so that is the driver change that breaks on recent RHEL 8.6. 

```
--- core/rtw_xmit.c.orig	2022-10-23 16:07:50.779073609 -0400
+++ core/rtw_xmit.c	2022-10-23 16:08:14.432451815 -0400
@@ -5159,12 +5159,14 @@
 	if (IS_CH_WAITING(adapter_to_rfctl(padapter)))
 		return -1;
 
+#if !defined(RHEL8)
 	if ((rtw_linked_check(padapter) == _FALSE)
 #ifdef CONFIG_LAYER2_ROAMING
 		&&(!padapter->mlmepriv.roam_network)
 #endif
 	   )
 		return -1;
+#endif
 
 	if (start == 0)
 		start = rtw_get_current_time();
--- core/rtw_recv.c.orig	2022-10-23 16:16:10.254059976 -0400
+++ core/rtw_recv.c	2022-10-23 16:16:23.012263974 -0400
@@ -4754,8 +4754,10 @@
 	if (MLME_IS_AP(adapter))
 		return 0;
 
+#if !defined(RHEL8)
 	if (rtw_linked_check(adapter) == _FALSE)
 		return 0;
+#endif
 
 	return 1;
 }

```

morrownr, 2022-10-23:
@tmcqueen-materials 

Do you want to submit that last fix as a PR or do you want me to get it onboard?

For my understanding, so that I can add something to a FAQ for future RHEL users, ... well, heck, can I get you to write a short note to RHEL users explained what you had to do for 8.6?

Nick

tmcqueen-materials, 2022-10-23:
@morrownr  PR created! Please tweak and edit as you see fit. I've only tested in Ad Hoc and Station mode. I suspect AP mode will be quite broken.

PR #91 (2022-10-31): Add hack to support EL 8.6 kernels (4.18.0-372)
...by ignoring the link state when transmitting and receiving frames (matching the behavior of ancient drivers for other chipsets that work on EL8.6). Fixes #90.
closed 2022-10-31 by morrownr