morrownr/88x2bu-20210702 #93: Error when installing

JoseDzRFP, 2022-11-05:
Error:
```
pi@raspberrypi ~/src/88x2bu-20210702 $ sudo ./install-driver.sh
Running install-driver.sh version 20221007
4.1.7-v7+
armv7l
Starting installation...
Copying rtw88_8822bu.conf to: /etc/modprobe.d
Copying 88x2bu.conf to: /etc/modprobe.d
The non-dkms installation routines are in use.
/bin/sh: 1: bc: not found
#make -C /lib/modules/4.1.7-v7+/build M=/home/pi/src/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
/bin/sh: 1: bc: not found
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.1.7-v7+/build M=/home/pi/src/88x2bu-20210702  modules
make[1]: *** /lib/modules/4.1.7-v7+/build: No such file or directory.  Stop.
Makefile:2503: recipe for target 'modules' failed
make: *** [modules] Error 2
An error occurred. Error = 2
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver-no-dkms.sh

morrownr, 2022-11-05:
Hi @JoseDzRFP 

> /bin/sh: 1: bc: not found

Did you run installation Step 3? bc should be installed if you did.

Is there a reason you are running a version of the RasPiOS that is using kernel 4.1? I don't test these drivers with anything older than 4.19. 4.1 is 7+ years old and went EOL in 2018. The repos are probably not there for this old unsupported version of RasPiOS so my strong recommendation is to update to something modern.

JoseDzRFP, 2022-11-06:
I just did what you said and the installation worked correctly, however when i plug in the adapter (Meross AC600 mwa255) nothing happens, no LED and no wireless interfaces found, am I supposed to do anything else or is the adapter not compatible?

morrownr, 2022-11-07:
> however when i plug in the adapter (Meross AC600 mwa255)

This driver is not for AC600 adapters so we need to get you on the right driver.

Post the results of the following:

$ lsusb

JoseDzRFP, 2022-11-07:
Omg I thought this was the correct driver, this is what appears when executing $ lsusb :
 ´´´jose@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 045e:0800 Microsoft Corp. Wireless keyboard (All-in-One-Media)
Bus 001 Device 006: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-11-07:
> Bus 001 Device 006: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter

The rtl8811au chipset uses the 8821au driver:

https://github.com/morrownr/8821au-20210708

Before installing the 8821au driver, recommend you go into the directory for the driver you have installed now and run:

$ sudo ./remove-driver.sh

... and then delete the directory. That should clean that driver out of your system.

Also, FYI: The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

There is a lot of good information there for those of us that use usb wifi adapters. Item 5 gives information on modern dual band Realtek chipset based adapters. It also gives links to the drivers that are located and supported here at this site,. Item 1 is a good read to give some background on usb wifi adapter use on Linux and item 2 is a long list of adapters that use in-kernel drivers.

Hope this helps. Let me know how it goes.

Nick

JoseDzRFP, 2022-11-07:
It worked! I installed the new drivers and after the las reboot I pluged in the adapter and it worked right away, thank you so much for helping me, you are great!