morrownr/88x2bu-20210702 #94: error when compiling

aini6658, 2022-11-07:
【in terminal】
'make' -j8 KVER=5.10.110 KSRC=/lib/modules/5.10.110/build..................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.13.1 not found
Error! Bad return status for module build on kernel: 5.10.110 (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.

【in /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log】
ERROR: modpost: module 88x2bu uses symbol kernel_write from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
ERROR: modpost: module 88x2bu uses symbol kernel_read from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
ERROR: modpost: module 88x2bu uses symbol filp_open from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.

aini6658, 2022-11-07:
https://www.androidpimp.com/embedded/nanopi-r6s/

an embedded board

morrownr, 2022-11-07:
Copy All. This is likely to be a challenge.

What distro and version are you using?

Did you run the `./ARM64_RPI.sh` script during the installation steps?

aini6658, 2022-11-08:
Running install-driver.sh version 20221007
5.10.110
aarch64
Starting installation...
Copying rtw88_8822bu.conf to: /etc/modprobe.d
Copying 88x2bu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=5.10.110 KSRC=/lib/modules/5.10.110/build..................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.13.1 not found
Error! Bad return status for module build on kernel: 5.10.110 (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

aini6658, 2022-11-08:
Ubuntu22.04 LTS/jammy

morrownr, 2022-11-09:
@aini6658 

Sorry for the delay. Too many irons in the fire.

So Ubuntu 22.04 on a ARM64 (aarch64) rk3588 processor.

The only thing I am almost certain about is that there is something not right about the dev environment. What the problem could be is not clear to me. It could be that the header files are messed up or it could be that your system has the wrong header files or even somthing else.

Since I have almost a zero chance of reproducing the issue since I do not have the hardware, you have to take charge of the investigation. I'll need information to help. Let;s try this first:

Go into the driver directory and run:

$ sudo ./remove-driver.sh
$ make clean
$ make

Post the output of `make`.

A follow thing to try:

$ sudo ./remove-driver.sh

Now go through each step of the installation again. I'm particularly interested to see if there is something that looks wrong with Steps 2 and 3.

Also, you did not answer my question above about whether you ran: `./ARM64_RPI.sh`

Cheers

hakaishi, 2022-11-24:
I get this too. My solution was to do the following.
1. uninstall
2. reboot
3. delete local git repo
4. clone
5. install

After doing a new clone, it built and installed without a problem.

closed 2023-03-01 by morrownr

huynguyen10-rvc, 2023-09-18:
Hi, just for anyone reading this later, I have resolved this by adding
```diff
diff --git a/os_dep/linux/os_intfs.c b/os_dep/linux/os_intfs.c
index 7baa1d6..68bee7e 100644
--- a/os_dep/linux/os_intfs.c
+++ b/os_dep/linux/os_intfs.c
@@ -21,6 +21,7 @@ MODULE_LICENSE("GPL");
 MODULE_DESCRIPTION("Realtek Wireless Lan Driver");
 MODULE_AUTHOR("Realtek Semiconductor Corp.");
 MODULE_VERSION(DRIVERVERSION);
+MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
 
 /* module param defaults */
 int rtw_chip_version = 0x00;
-- 
```
My reference: [Symbol Namespaces - The Linux Kernel documentation](https://www.kernel.org/doc/html/v5.7/core-api/symbol-namespaces.html)
Apparently, this is an Android thing: https://android.googlesource.com/kernel/common/+/refs/heads/android13-5.10/fs/Makefile#9