morrownr/88x2bu-20210702 #95: a user need support CSL-Computer wireless adapter

ohquait, 2022-11-20:
$ dmesg
[34001.333866] usb 1-8: new high-speed USB device number 14 using xhci_hcd
[34001.562475] usb 1-8: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[34001.562483] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[34001.562485] usb 1-8: Product: 802.11ac NIC
[34001.562487] usb 1-8: Manufacturer: Realtek
[34001.562489] usb 1-8: SerialNumber: 123456

have anyone here a other device with idProduct: b812 and idVendor 0bda (Realtek) ?
just a win10 Driver is published on the csl-computer support side

morrownr, 2022-11-20:
Hi @ohquait 

I have an adapter that has the same device id. What are you trying to do?

ohquait, 2022-11-21:
That's good,
i will recive the dongle in few day's and then i can check and post more about this thing,
but i believe this will a not real good firmware or mismatches in it .... i must wait for and then 
i can writing more exactly what's wrong....

ohquait, 2022-11-26:
Dear @morrownr ,
i hope the info's can help for other user's with this product:  
i have no issue with the wifi-device, just wondering about no MAC easy to find,
but nm-applet or networkmanager provides connections-informations and then i setting the 
published MAC address into router to allowing connecting this drive allways....
It's payed as CSL WIRELESS ADAPTER (CSL Computer )...... IEEE802.11 a/b/g/n/ac  2.4 & 5 GHz  USB3-Interface 
the linux drivers shipped with it are too old 2017, 
**your 88x2bu works with it !**!!
the device is:

> Bus 001 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 RTL88x2bu [AC1200 Techkey]
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac NIC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

the ID's 0bda:b812 is stored in realtek's source usb_intf.c as:

>#ifdef CONFIG_RTL8822B
	/*=== Realtek demoboard ===*/
	{USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0xB82C, 0xff, 0xff, 0xff), .driver_info = RTL8822B}, /* Default ID for USB multi-function */
	{USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, **0xB812**, 0xff, 0xff, 0xff), .driver_info = RTL8822B}, /* Default ID for USB Single-function, WiFi only */
	{USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0xB81A, 0xff, 0xff, 0xff), .driver_info = RTL8822B}, /* Default ID */
	/*=== Customer ID ===*/

that's all
ah, and it use the follow:
in
/etc/modprobe.d/rtl88x2bu.conf
options rtl88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1

closed 2022-11-26 by ohquait

morrownr, 2022-11-27:
> /etc/modprobe.d/rtl88x2bu.conf

The only thing that should be in the above file is:

blacklist rtw88_8822bu

The reason is that the only purpose of rtl88x2bu.conf is to blacklist an in-kernel driver that some distros have added. The driver is not mainline yet. I have had users report problems when they use a distros that has added this in-kernel driver so I took action by adding rtl88x2bu.conf to deactivate the in-kernel driver.

The name of the driver that this repo produces is 88x2bu.ko so the module parameters you show belong in:

/etc/modprobe.d/88x2bu.conf

I have a little script that makes sure you edit the correct file:

sudo ./edit-options.sh

`nano` should be installed.

> rtw_vht_enable=2

That setting should only be 2 if you are using AP mode. Are you? For everything else, a setting of 1 is best.

> the linux drivers shipped with it are too old 2017

Item 1 on the Main Menu addresses this problem:

https://github.com/morrownr/USB-WiFi

Regards

ohquait, 2022-11-27:
ah YES, i edited the driver-makefile to build module with name rtl88x2bu, 
that's a story of few years in PCLinuxOS, and there are not drivers with usb from rtw88, just the pci(e) 
connected drivers, ...  i know what you mean, but i going a other way for PCLinuxOS to support the old
names and so i renamed the module, but since we use your work, all is better with the drivers and
for that all i say BIG THANK YOU !!!

ohquait, 2022-12-28:
It's my pleasure to find the latest drivers here and package them to RPMs!
**Many thanks to everyone who is busy here!**
PCLinuxOS users love YOU!
In the meantime, drivers from the current repos are running from here.

**Best wishes for the next year!**

morrownr, 2022-12-29:
FYI: Myself and some others have been work on new code for the 8821cu driver. It should be released in the next few weeks. It has big changes to the scripts and docs as well as a few bugs fixed. We are going to flow the changes to the scripts and docs to the other drivers as time passes.

I'm not sure how you are packaging the driver but you need to be aware that using the 88x2bu.conf file for this driver and the 8821cu.conf file for the 8821cu driver is very important as the in-kernel drivers are coming and will conflict without those files installed to blacklist the in-kernel drivers.

> i edited the driver-makefile to build module with name rtl88x2bu

> package them to RPMs!

You are certainly welcome to do the above but keep in mind that you own the tech support if you do this. I've worked on these drivers for a few years and have had many requests for deb and rpm files but once I explain the implications of that, people usually go "oh, I didn't know that." My opinion is that the best long term solution is just to post links to the drivers so that people have good access to the documentation prior to installing the driver. 

Nick

ohquait, 2022-12-30:
renaming the module to rtl.... instead of .... for in kernel-drivers is
what i doing since many years, and that is the best way to
solve conflicts between drivers from kernel and 3rd-party drivers,

Would you have master-specfile and a master-makefile.patch and a
step-by-step-to-do
for RPM-packagers on your site? as Example for RPM-Builders/Packagers

morrownr, 2022-12-30:
Sir,

> renaming the module to rtl.... instead of .... for in kernel-drivers is
what i doing since many years, and that is the best way to
solve conflicts between drivers from kernel and 3rd-party drivers,

That does not solve conflicts, it creates them.

> Would you have master-specfile and a master-makefile.patch and a
step-by-step-to-do for RPM-packagers on your site?

Using package mangers to install out of kernel drivers is not a good solution unless there is someone that is working full time to repackage the drivers and add them to the system of automatic updates the distro has as they are constantly updated. There are many other reasons why I do not consider each distro making its own package to be of benefit to users. In fact, when i see that a user that posts a problem here or at one of my other repos is using a distro specific package, I opten turn them around and send them to the packager for support. I've had PClinuxOS users show up reporting problems and more often than not, the driver they installed via rpm has become dated and stopped working so they decided to install using the instructions here. How does that end? Badly. Why? Because the rpm packaged version is using a different name and is not detected and removed by my installation program.

My opinion is that the single best thing that distros can do regarding usb wifi adapters is make they best information available to their users and check periodically that the information remains correct but stay out of the business of packaging out-of-kernel drivers. It would be nice if the README for the drivers I have up had good information for PClinuxOS as it does for many other distros but it does not. The documentation is actually very important as there are many configuration options for various use cases and users may not pick up on that if they are installing via a package.

Does the package made available to PClinuxOS users install using dkms or manual type of installation? The reason I want to know is so that I can add code to my installation and removal scripts to rip out previous package installations so things don't blow as they will with 2 drivers for the same hardware are installed.

I am trying to support all distros that can be supported. I see a lot of things that distros do that make it hard to support them and that, in turn, hurts the user experience.

Maybe we can work together to discover what is of benefit to PClinuxOS users? I've been using Linux since 1994. Our collective experience could make things better if you want to work together. Let me know.

Nick

ohquait, 2022-12-31:
Hello Nick,
basically you are right of course.
I would prefer not to build the packages, but then I would have to hope
that the various git repos
for drivers always have a TAG or BRANCH in the name with a clear reference
to the last known good kernel.
An example, from another repo:
rtl8188fu this driver can currently only be used up to kernel 6.0.x
compile, with the update to Kernel-6.1.1 the code collides with changes in
the Linux headers and xxx.c files.
For the user this means: he has to be smart and keep the latest working
kernel installed so that he can use the wifi stick with the rtl8188fu chip
and driver. If he needs the functions from kernel 6.1.1, he reboots into
this kernel 6.1.1 but has to do without the rtl8188fu there.
And that's exactly the advantage of PCLinuxOS with the old DKMS system with
full RPM support. "Stupid only" if the user has no other network connection
than the rtl8188fu, then he can namely in the worst case
((if he manually selected the rpm in Synaptic to delete))
don't just download the working old RPM fresh from the apt repo.
But for that I want to integrate the backup strategy into my rpms,
so that in case of not compiling and deleting the driver
the user in an "offline repo" under
/usr/share/3rd-party-drivers/....
always its last working driver rpm to install with:
# apt-get install dkms-rtl8188fu-1.1-2pclos2022.x86_64.rpm
or with Synaptic PackageManager
can reinstall.
Renaming the individual modules is an easy way to avoid overwriting or
unintentionally reloading the original In-Kernel_Module, also installing
under:
/lib/modules/6.1.1-pclos2/kernel/3rdparty/rtl8188fu/ rtl8188fu.ko.xz
It doesn't matter to the kernel where it loads a module from, it doesn't
matter what it's called, the only important thing is that when the
individual module is loaded, the relevant kernel-in module is unloaded from
the kernel and possibly the options for the individual module its conf file
under
/etc/modprobe.d/rtl8188fu.conf
be implemented when loading.
Here I see it as an advantage that PCLinuxOS is configured more like a UNIX
with sysVinit than a systemd.../or whatever they are called.
But "YES" :
The effort is quite large for me, since I still have to earn my bread and
butter in a craftsman's profession. So I can only find the time to look for
updates in the git-repos at relatively regular intervals on a weekly basis.
But please don't support PCLinuxOS!
Feel free to post a request for this in your repos,
""...especially for PCLinuxOS users:
contact "onkelho" with a PM((personalMessage)) in the forum at:
https://www.pclinuxos.com/forum/index.php
...""
because I am dependent on the information and error messages there and in
PCLinuxOS, which unfortunately are not posted in sufficient quantity or
sometimes do not appear due to linguistic misunderstandings.
Everything else gladly, I would be very happy if this for one
saved you time with the pclos-issues, and so I'm able
to improve my support.

So much to write in google translator, man oh man...

So again, many THANKS to everyone who is doing this really valuable work
with YOU!
A "HAPPY NEW YEAR" everyone!

morrownr, 2023-01-01:
Happy New Year to you @ohquait 

It is going to take me some time to digest everything you said so expect another reply when I am able.

For today, let me leave you with this request:

Can you find a way to get the url for my Main Menu in front of PClinuxOS users?

https://github.com/morrownr/USB-WiFi

The primary goal of my effort is to get the best information regarding usb wifi adapters to all Linux users. Menu items 1 and 2 will add a lot of the knowledge of anyone who reads it no matter the distro they use. Many Linux users simply do know know the background or that there are MANY usb wifi adapters that use in-kernel drivers. In fact, the if a user wants the latest, highest performance chipset, the mt7921au is it and it is supported with a very good in-kernel driver. 2023 should see a flood of new adapters based on this chipset but adapters are already available. Nothing available from Realtek can meet the specs. The mt7921au can do WiFi 6 and WiFi 6e as well as bluetooth 5.2. I have tested it against other options that are available in WiFi 5 mode and it smokes the compention:

https://github.com/morrownr/USB-WiFi/issues/139

Take a look at my Main Menu and the information it contains. I think you will agree that this information would be very useful to PClinuxOS users.

Nick

ohquait, 2023-01-03:
> morrownr/USB-WiFi#139 <https://github.com/morrownr/USB-WiFi/issues/139>

Hey Nick,
i will send a copy of your e-Mail to the Owner of PCLinuxOS - Texstar - he
must say "OKAY" and then it comes on the Forum-site
under /NETWORKING,  or i will also become a Driver-section on this forum,,,
if he will nothing, than i create the Link to Your's on my site qua-it.org
in the default opening site NEWS...
Will that be OKAY for You?

morrownr, 2023-01-07:
Hey @ohquait 

Sorry for the delay. I've been busy and am behind on checking messages.

>  Will that be OKAY for You?

Sure. The link to this site that works best is the link to the Main Menu:

https://github.com/morrownr/USB-WiFi

If there is interest in adding PClinuxOS info to the README.md in each of the Realtek driver repos, that is fine. Let me know.

The new 8821cu driver is getting close to being ready. It is a really good driver. It even supports MU-MIMO. Would you like access to the repo? It is still private for testing for now but if you are interested, I can put you in.

Nick