morrownr/88x2bu-20210702 #97: Error installing: make error 2

beaver700nh, 2022-11-27:
```
Running install-driver.sh version 20221007
5.15.76-v7l+
armv7l
Starting installation...
Copying rtw88_8822bu.conf to: /etc/modprobe.d
Copying 88x2bu.conf to: /etc/modprobe.d
The non-dkms installation routines are in use.
#make -C /lib/modules/5.15.76-v7l+/build M=/home/pi/Downloads/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.15.76-v7l+/build M=/home/pi/Downloads/88x2bu-20210702  modules
make[1]: Entering directory '/usr/lib/modules/5.15.76-v7l+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.15.76-v7l+/build'
make: *** [Makefile:2503: modules] Error 2
An error occurred. Error = 2
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver-no-dkms.sh
```

Also, there is no such file `remove-driver-no-dkms.sh`

morrownr, 2022-11-27:
Hi @beaver700nh 

> Also, there is no such file remove-driver-no-dkms.sh

This is a typo that has been corrected and merged. You can update the source on your system with:

$ git pull

Regarding the error: My guess, since you did not specify, is that the distro is the 32 bit RasPiOS running a RasPi something. It would help if I knew what you are using.

I have doubt about installation step 3 as the installation script is running the non-dkms routines, which would be unusual for the RasPiOS. What actually killed the installation is either messed up or missing headers files.

kalmufti, 2022-12-02:
I have the same issue. You can easily replicate my setup. I'm running the standard Alpine Linux iso in a vm with mostly default ```setup-alpine``` and ```sys``` instal path. Would be great to get this driver working on Alpine.  

```bash
localhost:~/88x2bu-20210702# ./install-driver.sh
Running install-driver.sh version 20221126
5.15.80-0-lts
x86_64
Starting installation...
Copying rtw88_8822bu.conf to: /etc/modprobe.d
Copying 88x2bu.conf to: /etc/modprobe.d
The non-dkms installation routines are in use.
#make -C /lib/modules/5.15.80-0-lts/build M=/root/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.80-0-lts/build M=/root/88x2bu-20210702  modules
make[1]: Entering directory '/lib/modules/5.15.80-0-lts/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/5.15.80-0-lts/build'
make: *** [Makefile:2503: modules] Error 2
An error occurred. Error = 2
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```
```bash
localhost:~/88x2bu-20210702# uname -a
Linux localhost 5.15.80-0-lts #1-Alpine SMP Tue, 29 Nov 2022 08:08:44 +0000 x86_64 Linux
```

morrownr, 2022-12-02:
Hi @kalmufti

This driver can run on Alpine. There is something that has not been installed via installation step 3. I had the info for Alpine in step 3 but at some point it fell out. You or I can search the issues here and in the other repos for `alpine` and we will find it. 

I am neck deep on high priority work right now so if you find it and get this going, let me know what needs to be added into the README so this does not happen again. If you do not find it, check back with me and I'll look.

Nick

kalmufti, 2022-12-02:
Hi @morrownr

Sorry I deleted my initial comment and opened a new issue with more info.

I was able to build it, will update my findings on that issue I just need to make sure I can reproduce my process successfully, so will look into it next week.

closed 2023-03-01 by morrownr