morrownr/88x2bu-20210702 #98: Error won't build on Alpine Linux

kalmufti, 2022-12-02:
I'm running the standard Alpine Linux iso in a vm with mostly default ```setup-alpine``` and ```sys``` install path. Would be great to get this driver working on Alpine. Initially it was complaining that the ```build``` dir in ```'/lib/modules/5.15.80-0-lts/build``` don't exist so I've manually added it.

There are no ```dkms``` on Alpine but there is ```akms```

```bash
localhost:~# uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.15.80-0-lts x86_64
EFI variables are not supported on this system
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 0bda:b812
Bus 002 Device 001: ID 1d6b:0003
-ash: dkms: not found
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

```bash
localhost:~# akms status
akms: No modules found
```

```bash
localhost:~/88x2bu-20210702# ./install-driver.sh
Running install-driver.sh version 20221126
5.15.80-0-lts
x86_64
Starting installation...
Copying rtw88_8822bu.conf to: /etc/modprobe.d
Copying 88x2bu.conf to: /etc/modprobe.d
The non-dkms installation routines are in use.
#make -C /lib/modules/5.15.80-0-lts/build M=/root/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.80-0-lts/build M=/root/88x2bu-20210702  modules
make[1]: Entering directory '/lib/modules/5.15.80-0-lts/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/5.15.80-0-lts/build'
make: *** [Makefile:2503: modules] Error 2
An error occurred. Error = 2
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
localhost:~/88x2bu-20210702# ./remove-driver.sh
Running remove-driver.sh version 20221007
Starting removal...
./remove-driver.sh: line 53: dkms: command not found
An error occurred. dkms remove error = 127
Please report this error.
```

morrownr, 2022-12-04:
Hi @kalmufti 

> There are no dkms on Alpine but there is akms

I understand that some distros do not support dkms. The installation script detects if dkms is installed and if not, it uses a more basic installation.

Can I get you to tell me what you installed for the dev environment in installation step 3?

There is an issue with the development environment.

kalmufti, 2022-12-05:
Hi @morrownr,

The hint that helped me figure this out was the missing build directory as I mentioned.
Now that I can successfully reproduce the build I can safely say the missing requirement is the ```linux-lts-dev``` package.
On a vanilla Alpine install, users will just need to add the following:

```bash
apk add linux-lts-dev make gcc
```

Thanks for all the work you put into this!
\- Khalid

morrownr, 2022-12-05:
Hi @kalmufti 

Thanks for the info. I added the following the installations step 3 section so hopefully future Alpine users will know:

- Option for Alpine

```
sudo apk add linux-lts-dev make gcc
```

If you want to tell me what step 2 should have for updating and upgrading, I will put it also.

Nick

closed 2023-03-01 by morrownr