morrownr/88x2bu-20210702 #100: macaddr_acl=2 and external radius server issues

frankerooney, 2022-12-06:
Hi,
I'm using ubuntu 22.04, kernel 5.15.0-56, and a "AC1200 Techkey" usb dongle, usb id 0bda:b812

It's been in use with your repo driver for a fair while in AP mode with hostapd, and seems very solid now using 5 ghz, giving a decent transfer rate and support for a good number of stations.

I'd like to try and use Freeradius for mac authentication (to begin with - 802.1x later probably). 
My issue is when I set macaddr_acl=2 in the hostapd.conf, it simply doesn't reference the external radius server. If I simply change the wireless device to an older ralink one, using a very similar hostapd.conf, it works fine. 
Accounting does work with the 88x2bu driver, just not authentication.

So does the driver support external radius server reference? I read somewhere that some drivers lack the acl support for this.

Thanks

morrownr, 2022-12-06:
Hi @frankerooney 

> So does the driver support external radius server reference?

I have no idea. That is not something I have had to deal with so I simply don't know. And I am neck deep in work so I really don't have time to figure it out right now.

Here are a couple of things while you are help that might be of interest:

https://github.com/morrownr/USB-WiFi

That link is to the Main Menu of this site. Menu item 9 is about building an AP with usb wifi adapters and it contains a lot of adapter specific info that might help you.

Also, in the Main Menu, you might post an issue there as it gets more traffic than here and someone that knows might see it.

Nick

frankerooney, 2022-12-06:
Thanks Nick, I'll check out the link.
Cheers for the reply

closed 2023-03-01 by morrownr