morrownr/88x2bu-20210702 #101: BroTrend AC1200 is not recognized anymore / reinstallation fails

Npaffen, 2022-12-06:
For almost 1 1/2 week I had no issues using this driver to get my BroTrend AC1200 recognized by this driver. Suddenly it failed after a reboot. I reinstalled everything carefully deleting all dependencies. 

Your script failled with :
```
remove-driver.sh: 26: [[: not found
Running remove-driver.sh version 20221204
remove-driver.sh: 56: [[: not found
Linux Kernel=5.15.0-56-generic
CPU Architecture=x86_64
remove-driver.sh: 83: Syntax error: word unexpected (expecting ")")
`
Therefore I followed carefully [this guide](https://askubuntu.com/questions/1182236/how-to-properly-remove-kernel-modules-for-rtl88x2bu-in-ubuntu-mate-19-04).

Then rebooted and used your `install-driver.sh`. The output was the following:
sudo sh install-driver.sh 
install-driver.sh: 26: [[: not found
Running install-driver.sh version 20221204
install-driver.sh: 92: [[: not found
Linux Kernel=5.15.0-56-generic
CPU Architecture=x86_64
gcc --version=gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Installing rtw88_8822bu.conf to: /etc/modprobe.d
Installing 88x2bu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
install-driver.sh: 173: [[: not found
The driver was added successfully.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.0-56-generic KSRC=/lib/modules/5.15.0-56-generic/build.................
Signing module:
 - /var/lib/dkms/rtl88x2bu/5.13.1/5.15.0-56-generic/x86_64/module/88x2bu.ko
EFI variables are not supported on this system
/sys/firmware/efi/efivars not found, aborting.
cleaning build area...
install-driver.sh: 195: [[: not found
The driver was built successfully.

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-56-generic/updates/dkms/

depmod...
install-driver.sh: 210: [[: not found
The driver was installed successfully.
install-driver.sh: 234: read: Illegal option -n
install-driver.sh: 235: [[: not found
install-driver.sh: 240: read: Illegal option -n
install-driver.sh: 241: [[: not found
```
Surprisingly `lssub` still recognizes the adapter:
```
lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 007: ID 2717:ff88 Xiaomi Inc. Mi/Redmi series (RNDIS + ADB)
Bus 003 Device 005: ID 1ea7:0066 SHARKOON Technologies GmbH [Mediatrack Edge Mini Keyboard]
Bus 003 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 003 Device 002: ID 1a2c:0e24 China Resource Semico Co., Ltd USB Keyboard
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Can anyone help?

Npaffen, 2022-12-06:
Well...
I did so and this command rebooted my PC without any warning. Anyway, thanks for your help. Before the reboot I saw that `git pull` reported that the directory of the driver is up to date before starting with the deinstallation and installation. Still the problem is the same. The adapter is not recognized by the kernel? As you might already noticed I'm not extremly familiar with UNIX systems.

morrownr, 2022-12-06:
I will try to look at this later today. I really need more info. There is a pinned message at the top of the issues page with a title of something like Problem Reporting Guidelines. Read that message and post back.

Nick

morrownr, 2022-12-07:
Hi @Npaffen 

> Sry for beeing part of the large share of people who write before they read D:

As much as I try to learn how to read minds, it still is not working. I'm working hard behind the scenes to make using this and the other drivers here as easy to install as possible but it is a challenge with the various distros doing different things.

First, let me recommend you do the following:

```
git pull
sudo ./remove-driver.sh
sudo ./install-driver.sh
```

That will make sure any recent changes are in your local repo and then we can see where you are. 

I tested the driver on Linux Mint 21 just a few minutes ago:

```
$ sudo ./install-driver.sh
Running install-driver.sh version 20221205
Linux Kernel=6.1.0-060100rc5-generic
CPU Architecture=x86_64
gcc --version=gcc (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
Country Code==country US
Installing rtw88_8822bu.conf to: /etc/modprobe.d
Installing 88x2bu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added successfully.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=6.1.0-060100rc5-generic KSRC=/lib/modules/6.1.0-060100rc5-generic/build.....................
Signing module:
 - /var/lib/dkms/rtl88x2bu/5.13.1/6.1.0-060100rc5-generic/x86_64/module/88x2bu.ko
Secure Boot not enabled on this system.
cleaning build area...
The driver was built successfully.

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-060100rc5-generic/updates/dkms/

depmod...
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] y
Do you want to reboot now? (recommended) [y/N] 
```

That is what you should be seeing.

Nick

Npaffen, 2022-12-07:
```
Already up to date.
Running remove-driver.sh version 20221204
Linux Kernel=5.15.0-56-generic
CPU Architecture=x86_64
Removing a dkms installation.
Removing rtw88_8822bu.conf from /etc/modprobe.d
Removing 88x2bu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl88x2bu-5.13.1
The driver was removed successfully.
You may now delete the driver directory if desired.
Do you want to reboot now? (recommended) [y/N] N
Running install-driver.sh version 20221205
Linux Kernel=5.15.0-56-generic
CPU Architecture=x86_64
gcc --version=gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Country Code==country 00: DFS-UNSET
Installing rtw88_8822bu.conf to: /etc/modprobe.d
Installing 88x2bu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added successfully.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.0-56-generic KSRC=/lib/modules/5.15.0-56-generic/build......................
Signing module:
 - /var/lib/dkms/rtl88x2bu/5.13.1/5.15.0-56-generic/x86_64/module/88x2bu.ko
EFI variables are not supported on this system
/sys/firmware/efi/efivars not found, aborting.
cleaning build area...
The driver was built successfully.

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-56-generic/updates/dkms/

depmod....
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] >
Do you want to reboot now? (recommended) [y/N] N

```

The unsupported EFI variables are the only thing that is different in the building I guess. I can't restart right now since I download an important file I need for work tomorrow using USB-Tethering. But will reboot afterwards and report back. Compared to OP the installation looked much cleaner.

morrownr, 2022-12-08:
> Country Code==country 00: DFS-UNSET

That is a problem. Do you know how to set and check the country code?

> The unsupported EFI variables are the only thing that is different in the building I guess. 

That basically just means you are on a system that does not support secure boot so it is not an issue if that is correct.

Npaffen, 2022-12-08:
I thought this could by done by adjusting the `locale` file in /etc/default/`. But those locales are all correctly set. Does this has to be done inside the makefile?

morrownr, 2022-12-08:
> Does this has to be done inside the makefile?

No. I'm working on a document file to explain this but it is not ready yet. Do this:

$ iw reg set US

where US is your country code. If not US, tell me the country and I can tell you the code.

To check:

$ iw reg get

Let me know how that works.

Nick

Npaffen, 2022-12-09:
I did so and set my country code to DE which should be correct since my keyboard was set to DE...

I guess not. Now it seems that my kernel is going wild when I try to boot into MINT.

I only ran into module:x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000001b36095a, val fffffffc4285074

morrownr, 2022-12-09:
> Now it seems that my kernel is going wild when I try to boot into MINT.

> When I try to boot the only message that appears repeatedly: module:x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000001b36095a, val fffffffc4285074

Is this a fairly old system? I'm not knocking old systems. The system I am on right now is 11 years old and it is my main dev box. I'm seeing some things that could indicate a failure in some component. Were you seeing odd behavior or stabilitiy issue before you first installed this adapter?

> $ iw reg set US

I should have written:

$ sudo iw reg set US

Npaffen, 2022-12-09:
No the system is quite up to date.
From my edit in the  OP I'm running on MINT 21: 

> The name and version of the Linux distro you are using
> ```
> cat /etc/*-release
> DISTRIB_ID=LinuxMint
> DISTRIB_RELEASE=21
> DISTRIB_CODENAME=vanessa
> DISTRIB_DESCRIPTION="Linux Mint 21 Vanessa"
> NAME="Linux Mint"
> VERSION="21 (Vanessa)"
> ID=linuxmint
> ID_LIKE="ubuntu debian"
> PRETTY_NAME="Linux Mint 21"
> VERSION_ID="21"
> HOME_URL="https://www.linuxmint.com/"
> SUPPORT_URL="https://forums.linuxmint.com/"
> BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
> PRIVACY_POLICY_URL="https://www.linuxmint.com/"
> VERSION_CODENAME=vanessa
> UBUNTU_CODENAME=jammy
> cat: /etc/upstream-release: Is a directory
> ```

Npaffen, 2022-12-09:
I was able to use an older kernel to get back into my system but I guess my 5.16.x kernel is done. The error message looks like some hashsum in one of the header-files is corrupt? This probably only happens when you got your hands very dirty while editing files you should better never touch if you don't know exactly what your are doing.

Fun fact: On my 5.15 kernel the USB-Wifi dongle was recognized and usable immediatly.

morrownr, 2023-03-01:
@Npaffen 

I'm going to close this issue as it likely is no longer a problem.

> This probably only happens when you got your hands very dirty while editing files you should better never touch if you don't know exactly what your are doing.

This driver never modifies header files.

closed 2023-03-01 by morrownr