morrownr/88x2bu-20210702 #104: TP-Link Archer T4U ver.3 does not initialize?

jpap, 2022-12-08:
I have a [TP-Link Archer T4U ver.3](https://www.amazon.com/wireless-USB-WiFi-Adapter-PC/dp/B08KHV7H1S/ref=asc_df_B08KHV7H1S/) that was recently purchased.  It works 100% under Windows, and I am trying to get it working on Linux.

I've tried VMs on amd64 (Linux kernel 5.15) and arm64 (Debian; Linux kernel 5.10), and a bare metal armv6l (kernel 5.15; Raspberry Pi B+): they all give the same result shown below.

Just in case, I tried it with a [rtl8812au driver](https://github.com/aircrack-ng/rtl8812au), and the device didn't register with the driver when plugged in; so it almost surely appears to be the BU chipset?

<details>
<summary>A full driver debug log is here.  Perhaps the reason for the failure are the following two lines:
<block><pre>
[  +0.000607] RTW: ERROR [HALMAC][ERR]test mode!!
[  +0.000002] RTW: ERROR rtl8822b_power_on: Power ON Fail!!
</pre></block>
</summary>
<pre>
[  +4.363461] RTW: module init start
[  +0.000002] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[  +0.000001] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[  +0.000010] RTW: rtw_inetaddr_notifier_register
[  +0.000018] RTW: 
              usb_endpoint_descriptor(0):
[  +0.000001] RTW: bLength=7
[  +0.000000] RTW: bDescriptorType=5
[  +0.000000] RTW: bEndpointAddress=84
[  +0.000001] RTW: wMaxPacketSize=512
[  +0.000000] RTW: bInterval=0
[  +0.000001] RTW: RT_usb_endpoint_is_bulk_in = 4
[  +0.000000] RTW: 
              usb_endpoint_descriptor(1):
[  +0.000000] RTW: bLength=7
[  +0.000001] RTW: bDescriptorType=5
[  +0.000000] RTW: bEndpointAddress=5
[  +0.000000] RTW: wMaxPacketSize=512
[  +0.000001] RTW: bInterval=0
[  +0.000000] RTW: RT_usb_endpoint_is_bulk_out = 5
[  +0.000000] RTW: 
              usb_endpoint_descriptor(2):
[  +0.000001] RTW: bLength=7
[  +0.000000] RTW: bDescriptorType=5
[  +0.000000] RTW: bEndpointAddress=6
[  +0.000001] RTW: wMaxPacketSize=512
[  +0.000000] RTW: bInterval=0
[  +0.000000] RTW: RT_usb_endpoint_is_bulk_out = 6
[  +0.000001] RTW: 
              usb_endpoint_descriptor(3):
[  +0.000000] RTW: bLength=7
[  +0.000000] RTW: bDescriptorType=5
[  +0.000001] RTW: bEndpointAddress=87
[  +0.000000] RTW: wMaxPacketSize=64
[  +0.000000] RTW: bInterval=3
[  +0.000001] RTW: RT_usb_endpoint_is_int_in = 7, Interval = 3
[  +0.000000] RTW: 
              usb_endpoint_descriptor(4):
[  +0.000000] RTW: bLength=7
[  +0.000001] RTW: bDescriptorType=5
[  +0.000000] RTW: bEndpointAddress=8
[  +0.000000] RTW: wMaxPacketSize=512
[  +0.000001] RTW: bInterval=0
[  +0.000000] RTW: RT_usb_endpoint_is_bulk_out = 8
[  +0.000000] RTW: nr_endpoint=5, in_num=2, out_num=3

[  +0.000001] RTW: USB_SPEED_HIGH
[  +0.000001] RTW: CHIP TYPE: RTL8822B
[  +0.000018] RTW: [HALMAC]55772M
              HALMAC_MAJOR_VER = 1
              HALMAC_PROTOTYPE_VER = 6
              HALMAC_MINOR_VER = 6
              HALMAC_PATCH_VER = 23
[  +0.001397] RTW: [HALMAC][TRACE]halmac_init_adapter <===
[  +0.000001] RTW: [HALMAC][TRACE]phy_cfg_usb_8822b ===>
[  +0.000736] RTW: [HALMAC][TRACE]phy_cfg_usb_8822b <===
[... 451 lines cut ...]
        ether 00:1c:42:02:48:2f  txqueuelen 1000  (Ethernet)
        RX packets 1408  bytes 361880 (353.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 908  bytes 202225 (197.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx984827ebc5e9: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::9a48:27ff:feeb:c5e9  prefixlen 64  scopeid 0x20<link>
        ether 98:48:27:eb:c5:e9  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 12 overruns 0  c
```

Appreciate any advice. :)

jpap, 2022-12-08:
This is where the first failure happens:

https://github.com/morrownr/88x2bu-20210702/blob/2590672d717e2516dd2e96ed66f1037a6815bced/hal/halmac/halmac_88xx/halmac_8822b/halmac_init_8822b.c#L991-L994

jpap, 2022-12-08:
If I comment out line 993 above, it works! :D

morrownr, 2022-12-09:
Hi @jpap 

Interesting. This is a new one on me. Run it for a while with the line commented out and see what happens. I'd like to know.

By the way, the approval was given today for the new in-kernel driver for this chipset to be mainlined. I'm not sure if it will make it into 6.2 or 6.3 but there is something to be excited about going on. Imagine not having to compile a driver.

jpap, 2022-12-14:
It's been very stable.  I've been running it as a AP client for the past week 24/7 and have had no problems.  When the Raspberry Pi reboots, it always comes back up and associates.

morrownr, 2022-12-14:
Hi @jpap 

That is good news. I think I am going to borrow your fix above and see what happens on other adapters. In looking at the code, it does not seem necessary for that code to exist anyway but we'll see. Thanks.

Nick

jpap, 2022-12-14:
My guess is that the check is there to just make sure the adapter is responding to the commands issued to it before issuing more for that mode.

It might be that with my device they changed the register value, dropped it entirely (no other driver code reads or writes that register), or there’s a race condition for that value to be valid. I am getting a second T4U for another use case, now that it works with Linux, and will see if that exhibits the same behaviour.

If there’s time, I’ll add a sleep or delay to see if the problem is caused by a race, or confirm if the hardware registers were changed in that silicon revision.

Thank you otherwise for your maintenance work and quick responses to these issues. I would’ve had a paperweight on my hands otherwise!

morrownr, 2022-12-14:
@jpap 

Response much appreciated. I would appreciate information from your further testing so as to note it here. If you can't find anything and want to submit a PR, that would be fine.

> I am getting a second T4U for another use case, now that it works with Linux...

A couple of things to point out to you:

- The Main Menu to this site is:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 give a lot of information about wifi adapters and use on Linux. Keep in mind that there are adapters with the new mt7921au chipset available and it is WiFi 6e capable. It is very fast on 5 GHz in AC mode. Still testing WiFi 6 and WiFi 6e.

- A new series of patches has gone in and will be in kernel 6.2. It is a new in-kernel driver for this chipset.

I plan to test once 6.2 rc1 is released here in a couple of weeks. I don't know about the quality yet.

Nick

jpap, 2022-12-17:
My second device (the same brand/model) arrived by mail today... and it works without the above patch!

Given that my first device worked under Windows without any issues, I'm not convinced it is a hardware problem per-se.  It could be a different revision of the silicon (I haven't opened them up, because I don't want to destroy the bottom sticker and void the warranty), or maybe it is indeed a race condition in the driver.

Since a) my "buggy device" works fine for an extended period with the change outlined above, b) the register was not written to, and c) the register doesn't appear anywhere else in the driver, I'd be inclined to recommend making that change permanently for everyone.

If I get a chance, I'll add a spin/sleep to confirm or rule out the race condition idea; but I might not be able to do that for some time as that device is running remotely at present.

Please close this issue, along with a commit or not, as you feel.  I'd also be happy to submit a PR and do it that way if you prefer.

morrownr, 2022-12-17:
> Since a) my "buggy device" works fine for an extended period with the change outlined above, b) the register was not written to, and c) the register doesn't appear anywhere else in the driver, I'd be inclined to recommend making that change permanently for everyone.

No need to submit a PR. I've already made the change locally and will push it later today. Thanks for investigating.