morrownr/88x2bu-20210702 #107: A required package appears to not be installed. Please install the following package: mokutil

GenerallyClumsy, 2022-12-21:
Hi there, 

just a heads up for people with Rasbian lite they need to manually install mokutil. Probably worth adding a "sudo apt-get install mokutil" to  the instructions. https://github.com/morrownr/88x2bu-20210702

Thanks for the great work!

morrownr, 2022-12-21:
Hi @GenerallyClumsy 

Thanks for the feedback and glad the installation went well.

Myself and a few others are working to greatly enhance docs and scripts right now. Some of the features are already working their way into this repo. The issue you bring up with mokutil has already be reconsidered and is being handled a different way. There are other checking routines in place now that you may see at some point but we have to stay away from distro specific help like "sudo apt install mokutil" because that is Debian specific and would not work on Fedora, OpenSuSE, Arch and many more. All we can really do tell users that they need to install a package.

Let us know if you see any problems.

FYI: A new in-kernel driver for this chipset was up-streamed for kernel 6.2. I have not tested the quality yet. This driver is ready in that it blacklists the in-kernel driver when installed and reactivated the in-kernel driver when removed.

Nick

morrownr, 2022-12-22:
Hi @GenerallyClumsy 

I have now merged a major update to the scripts and docs. Could I get you to test to see if you see any problems:

$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh

Thanks

closed 2023-03-01 by morrownr