morrownr/88x2bu-20210702 #111: Seg fault when installing

iyush, 2023-01-09:
When executing the script:
```
aayush@dev-pc:~/88x2bu-20210702$ sudo ./install-driver.sh 
Script:  install-driver.sh version 20221228
Kernel:  5.15.0-57-generic
Arch:  x86_64
gcc:  gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Country:  country 00: DFS-UNSET
The Country Code may not be properly set.
File alpha-2_Country_Codes is located in the driver directory.
Please read and follow the directions in the file after installation.
SecureBoot disabled
Platform is in Setup Mode
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
                 /usr/src/rtl88x2bu-5.13.1

DKMS: add completed.
The driver was added to dkms successfully.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j16 KVER=5.15.0-57-generic KSRC=/lib/modules/5.15.0-57-generic/build.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.0-57-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 167.19 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

Logs:
```
DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.15.0-57-generic (x86_64)
Mo 9. Jan 11:44:12 CET 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-57-generic/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-57-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
...
...
...
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/btc/halbtc8822b2ant.o
In file included from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_conf.h:17,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_types.h:25,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/btc/mp_precomp.h:18,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/btc/halbtc8822b2ant.c:16:
/var/lib/dkms/rtl88x2bu/5.13.1/build/include/autoconf.h:181: warning: "CONFIG_RTW_SW_LED" redefined
  181 |  #define CONFIG_RTW_SW_LED
      | 
<command-line>: note: this is the location of the previous definition
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/platform/platform_ops.o
  LD [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.o
Segmentation fault (core dumped)
make[2]: *** [scripts/Makefile.build:494: /var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.o] Error 139
make[1]: *** [Makefile:1902: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-57-generic'
make: *** [Makefile:2506: modules] Error 2

```

morrownr, 2023-01-09:
Hi @iyush 

I just did a quick fresh installation test and was not able to duplicate this. Give me a chance to see if I can figure out what happened.

morrownr, 2023-01-09:
@iyush 

I just merged a lot of updates, including one that possibly could fix your issue. Please run the removal script, update your repo and install.

$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh

closed 2023-03-01 by morrownr