morrownr/88x2bu-20210702 #112: Unable to set channel when in monitor mode

camm73, 2023-01-09:
I've been using this driver with a 5.4 Linux kernel on hardware with a Wifi/BT module that uses the RTL8822BU chipset. I'm setting the interface to monitor mode using the `iw` CLI (netlink). Once in monitor mode, I try to set the channel also using `iw` (e.g. `iw dev wlan0 set channel 11 NOHT`); however, regardless of which channel I try to set, `iw dev wlan0 info` always shows the channel never changing from channel 1. It should be noted that I can successfully change other interface configuration parameters while in monitor mode such as `txpower`, for example—just not the channel.

When running in managed mode I've noticed that the channel successfully changes to the channel of the network it is connected to, so this issue seems to be limited to monitor mode. Do you have any ideas why this would be happening?

morrownr, 2023-01-09:
Hi @camm73 

> Do you have any ideas why this would be happening?

Let me investigate. In the mean time, let me point you to a repo I have that is about Monitor mode:

https://github.com/morrownr/Monitor_Mode

camm73, 2023-01-09:
Ran the script in that Monitor Mode repo as a sanity check, but I'm still seeing the same behavior where the channel does not get set properly and remains on channel 1

morrownr, 2023-01-10:
@camm73 

It appears to be fixed. I just merged a large patch. You will need to update the code on your system and reinstall:

$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh

Let me know.

camm73, 2023-01-11:
Just tested it out with the new changes and it's working great. Thanks so much for the help and the quick turnaround on that!

closed 2023-01-11 by camm73