morrownr/88x2bu-20210702 #113: Working on Arch Linux x86_64 Kernel 6.1.4-arch1-1

berylleigh, 2023-01-10:
Thank you so much for your work.  
Letting you know the driver works on Arch Linux x86_64 Kernel 6.1.4-arch1-1 with an "ac1300" usb adapter that shows up with lsusb with this device ID: 
0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

https://www.amazon.com.au/dp/B09JB1TXF9?psc=1&ref=ppx_yo2ov_dt_b_product_details
[image]

Thanks again!  Great work!

morrownr, 2023-01-14:
Hi @berylleigh 

Should I add this adapter to the Tested Hardware section?

Also, I merged a patch this morning that fixes beamforming support (MU-MIMO). Are you about to test if is working? If so...

install `wavemon`
start `wavemon` in a terminal
note the `link quality and signal strength` levels
start a download and check the `link quality and signal strength` levels again

Cheers,

Nick

berylleigh, 2023-01-16:
Yes - please add it to the tested hardware section.  It's impressive the driver supports beamforming - once again - great work!

morrownr, 2023-01-19:
@berylleigh 

I noticed the adapter is out of stock. I'd prefer to wait and see if it comes back in stock before posting.

Have you had a change to see if the beamforming is working for you? You router will have to support it also.