morrownr/88x2bu-20210702 #114: Error 6

hlinuxmintid, 2023-01-12:
: x86_64
: 5.15.86-1-lts
: gcc (GCC) 12.2.0
: dkms-3.0.9
rtl88x2bu/5.13.1, 5.15.86-1-lts, x86_64: built
rtl88x2bu/5.13.1.r179.a2ac3b2, 5.15.86-1-lts, x86_64: installed
rtl88x2bu/5.13.1.r179.a2ac3b2, 6.1.4-arch1-1, x86_64: installed (WARNING! Missing some built modules!)
rtl88x2bu/r142.549257e, 5.15.86-1-lts, x86_64: built
rtl88x2bu/r142.549257e, 6.1.4-arch1-1, x86_64: built (WARNING! Missing some built modules!)
Removing a dkms installation: rtl88x2bu
Module rtl88x2bu 5.13.1 is not installed for kernel 5.15.86-1-lts (x86_64). Skipping...
Deleting module rtl88x2bu-5.13.1 completely from the DKMS tree.
Removing 88x2bu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl88x2bu-5.13.1
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
Sign command: /usr/lib/modules/5.15.86-1-lts/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j4 KVER=5.15.86-1-lts KSRC=/lib/modules/5.15.86-1-lts/build.......................
Signing module /var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko
Cleaning build area...
The driver was built by dkms successfully.

88x2bu.ko.zst:
Running module version sanity check.
Module version v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b for 88x2bu.ko.zst
exactly matches what is already found in kernel 5.15.86-1-lts.
DKMS will not replace this module.
You may override by specifying --force.
Error! Installation aborted.
An error occurred. dkms install error:  6
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

hlinuxmintid, 2023-01-12:
Thanks Nick!  Unable to install it.  Enough is enough for me; I surrender.  Again, thanks you.

morrownr, 2023-01-13:
Compiling drivers is not for everyone. I recommend most folks go with adapters that in-kernel drivers so that it is plug and play.

https://github.com/morrownr/USB-WiFi

If you read 1 and 2, it might help.

closed 2023-03-01 by morrownr