morrownr/88x2bu-20210702 #116: Error 0

Krystjan30, 2023-01-16:
I need some help to install the driver for my edimax AC1200 USB 3.0 on Ubuntu 22.10
Here is the report...
Thanks alot!

# sh install-driver.sh 
install-driver.sh: 33: [[: not found
: install-driver.sh v20230109
: x86_64
: 5.19.0-29-generic
: gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
: SecureBoot disabled
install-driver.sh: 161: [[: not found
install-driver.sh: 175: [[: not found
install-driver.sh: 191: [[: not found
Installing 88x2bu.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.0-29-generic/build M=/home/db16/Téléchargements/88x2bu-20210702-main  modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.19.0-29-generic »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  You are using:           gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_cmd.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_security.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_debug.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_io.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_ioctl_query.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_ioctl_set.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_ieee80211.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_mlme.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_mlme_ext.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_mi.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_wlan_util.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_vht.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_pwrctrl.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_rf.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_chplan.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/monitor/rtw_radiotap.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_recv.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_sta_mgt.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_ap.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/wds/rtw_wds.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/mesh/rtw_mesh.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_xmit.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_p2p.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_rson.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_tdls.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_br_ext.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_iol.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_sreset.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_btcoex_wifionly.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_btcoex.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_beamforming.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_odm.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_rm.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_rm_fsm.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_ft.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_wnm.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_mbo.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_rm_util.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/efuse/rtw_efuse.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/rtw_roch.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/aes-internal.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/aes-internal-enc.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/aes-gcm.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/aes-ccm.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/aes-omac1.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/ccmp.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/core/crypto/gcmp.o
[... 115 lines cut ...]
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/halrf/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/halrf/rtl8822b/halrf_rfk_init_8822b.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/platform/platform_ops.o
  LD [M]  /home/db16/Téléchargements/88x2bu-20210702-main/88x2bu.o
  MODPOST /home/db16/Téléchargements/88x2bu-20210702-main/Module.symvers
  CC [M]  /home/db16/Téléchargements/88x2bu-20210702-main/88x2bu.mod.o
  LD [M]  /home/db16/Téléchargements/88x2bu-20210702-main/88x2bu.ko
  BTF [M] /home/db16/Téléchargements/88x2bu-20210702-main/88x2bu.ko
Skipping BTF generation for /home/db16/Téléchargements/88x2bu-20210702-main/88x2bu.ko due to unavailability of vmlinux
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.19.0-29-generic »
install-driver.sh: 231: [[: not found
install -p -m 644 88x2bu.ko  /lib/modules/5.19.0-29-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 5.19.0-29-generic
install-driver.sh: 248: [[: not found
An error occurred:  0
Please report this error.
Please copy all screen output and paste it into the problem report.

morrownr, 2023-01-16:
Hi @Krystjan30 

I test with bash, not sh. I need you to following the instructions carefully. First lets clean things up and then try again:

$ sudo ./remove-driver.sh
$ sudo ./install-driver.sh

Let me know how it goes.

Nick

Krystjan30, 2023-01-16:
Hi Nick, 
Thanks to your quick and efficient help, everything went very well this time. Thank you so much!
Krystjan

closed 2023-01-17 by Krystjan30

morrownr, 2023-01-18:
Glad it worked out. Stop by anytime.

Nick