morrownr/88x2bu-20210702 #118: AP mode doesn't have 5GHz

edwardwong80, 2023-01-21:
Thanks for the driver, in my x86 mini PC I installed this driver and client mode it's working nicely, though it's only USB2 mode (the manufacturer makes it in USB2 mode so the mod_switch option doesn't work).

I was trying to test the AP mode with Linux Mint, no matter what option I choose, the outcome is only 2.4GHz available (Yes it can share WiFi without problem but just no 5GHz), can you give me some direction to troubleshoot this?

morrownr, 2023-01-22:
Are you using Cinnamon?

Is your computer connected to the internet with ethernet?

edwardwong80, 2023-01-22:
Yes Linux Mint with Cinnamon, the other side I have Ethernet connection to
home internet and share it over WiFi to test AP mode but only 2.4GHz signal.

morrownr, 2023-01-22:
I have a reasonably detailed guide for setting up an AP:

https://github.com/morrownr/USB-WiFi

The guide is menu item 9. It is not specifically for setting up Linux Mint but does have information that might help you.

To your specific situation, when I hear 2.4 is working and 5 is not, I think country code.Post the results of:

$ iw reg get

If you need to set the country code:

$ sudo iw reg set US

If not in US, use the code for the country that you live in.

For me to help more, I would need the details of how you have AP mode setup.

Nick

edwardwong80, 2023-01-23:
Because of the machine having another onboard WiFi, so it's kind of messy, I switched to another one (an ARM board) with no WiFi.

Running Debian Bullseye, and here is the output of "iw reg get"

```
root@DietPi:~# iw reg get
global
country JP: DFS-JP
       (2402 - 2482 @ 40), (N/A, 20), (N/A)
       (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM
       (4910 - 4990 @ 40), (N/A, 23), (N/A)
       (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
       (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
       (5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
       (57000 - 66000 @ 2160), (N/A, 10), (N/A)
```

Now I set country code in 88x2bu.conf with this option string:

```
options 88x2bu rtw_country_code=JP rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_switch_usb_mode=0
```

Previously missed out an option in hostapd so can't see 5GHz, now I changed it back but I don't seem to be able to get proper bit rate at 5GHz (only 20M channel width probably the reason) , here is my demsg log when client joining.

```
[ 1183.121606] RTW: ============ STA [34:76:c5:85:02:88]  ===================
[ 1183.121647] RTW: mac_id : 3
[ 1183.121660] RTW: wireless_mode : 0x14
[ 1183.121673] RTW: mimo_type : 0
[ 1183.121684] RTW: static smps : N
[ 1183.121695] RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
[ 1183.121708] RTW: rate_id : 5
[ 1183.121720] RTW: rssi : -1 (%), rssi_level : 0
[ 1183.121734] RTW: is_support_sgi : N, is_vht_enable : Y
[ 1183.121747] RTW: disable_ra : N, disable_pt : N
[ 1183.121759] RTW: is_noisy : N
[ 1183.121770] RTW: txrx_state : 0
[ 1183.121782] RTW: curr_tx_rate : CCK_1M (L)
[ 1183.121795] RTW: curr_tx_bw : 20MHz
[ 1183.121806] RTW: curr_retry_ratio : 0
[ 1183.121818] RTW: ra_mask : 0x00000000000ffff0
[ 1183.124126] RTW: send eapol packet 1/4
[ 1183.198111] RTW: recv eapol packet 2/4
[ 1183.199192] RTW: send eapol packet 3/4
[ 1183.200754] RTW: recv eapol packet 4/4
[ 1183.205289] RTW: set pairwise key camid:3, addr:34:76:c5:85:02:88, kid:0, type:AES
```

Here is my hostapd.conf:

```
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
wpa_key_mgmt=WPA-PSK
beacon_int=100
ssid=raspi-webgui
channel=40
hw_mode=a
ieee80211n=1
wpa_passphrase=ChangeMe
interface=wlan0
wpa=2
wpa_pairwise=CCMP
country_code=JP
ignore_broadcast_ssid=0
ieee80211ac=1
wmm_enabled=1
require_vht=1
require_ht=1
```

iwconfig shows:

```
wlan0     IEEE 802.11AC  ESSID:"raspi-webgui"  Nickname:"WIFI@RTL88X2BU"
         Mode:Master  Frequency:5.2 GHz  Access Point: 1C:BF:CE:E4:1F:90   
         Bit Rate:174 Mb/s   Sensitivity:0/0  
         Retry:off   RTS thr:off   Fragment thr:off
         Encryption key:off
         Power Management:off
         Link Quality=21/100  Signal level=11/100  Noise level=0/100
         Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
         Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Any clue why there is such problem? The dongle supposed to be 2x2 which should show 866Mb/s

morrownr, 2023-01-23:
The following setting is more stable:

>rtw_switch_usb_mode=0

...but the following is what it will take to get the speed up:

rtw_switch_usb_mode=1

edwardwong80, 2023-01-24:
No it doesn't work, first of all I was told by manufacturer this is a USB 2.0 device (and I tried which the device couldn't even bring up after boot).

And from demsg it always says 20MHz channel which I believe this is a problem, and I found no way to make it a 80MHz one.

morrownr, 2023-01-24:
> No it doesn't work, first of all I was told by manufacturer this is a USB 2.0 device (and I tried which the device couldn't even bring up after boot).

Then the chipset is likely 8822bu which is the combo version which supports wifi and bt. That is fine, this driver supports that chipset. This has to do with USB3 interfering with bt.
 
> And from demsg it always says 20MHz channel which I believe this is a problem, and I found no way to make it a 80MHz one.

Let's make a few changes to hostapd.conf to get you to 80 MHz:

```
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
wpa_key_mgmt=WPA-PSK
ssid=raspi-webgui
# Channel
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# Country code
country_code=JP
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
hw_mode=a
wpa_passphrase=ChangeMe
interface=wlan0
wpa=2
wpa_pairwise=CCMP
ignore_broadcast_ssid=0
# you do not want the below line
#require_vht=1
# you do not want the below line
#require_ht=1

ieee80211n=1
wmm_enabled=1
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]

```

Okay. I think that should do it. The information was in the file at the link I posted earlier but I know it can be a challenge to pick the correct info out.

Let me know how it goes.

Nick

fakemanhk, 2023-01-25:
Oh cool, it works! Now I can see that's running at 80MHz channel width, I did dig into manpage of hostapd.conf however some of the parameters are difficult to understand.

Now I have to see how NanoPi folks managed to make this work on their customized FriendlyElec OpenWRT because using vanilla OpenWRT I have no way to make this dongle working, however they were able to do that. Anyway thanks for the information!

morrownr, 2023-01-26:
> because using vanilla OpenWRT I have no way to make this dongle working...

I'm using an adapter based on the mt7921au chipset with OpenWRT 22.03.3. The OpenWRT devs backported the in-kernel driver:

kmod-mt7921u

The adapter I use is the cf-961ax. It adds WiFi 6 to my wifi router. If you go to the Main Menu:

https://github.com/morrownr/USB-WiFi

Go to menu item 2.

fakemanhk, 2023-01-27:
Yes I saw your post on USB WiFi, planning to get it but now it's new year in China all shipment delaying.

I saw you mentioned that another dongle which needs to modify ID to make it work, may I know how to do that? I have a fairy old USB mem stick style mt7610 (or 7612) based 802.11ac WiFi (433+200 type), planned to use it with NanoPi R2S to make a travel router, however I have no idea why the USB ID is not correct so that I couldn't get driver to work under Linux (but in Windows I observed that it's using Ralink driver which means Mediatek now and it works there), so I would like to know how to implement this trick to make it work on my side.

morrownr, 2023-01-27:
> I have a fairy old USB mem stick style mt7610 (or 7612) based 802.11ac WiFi (433+200 type)

The first step is figuring out exactly what chipset and device ID this adapter has:

$ lsusb

Please post the results.

fakemanhk, 2023-01-28:
Here is the result of _lsusb -v_:

```

 Bus 001 Device 011: ID 04bb:0955 I-O Data Device, Inc. WHG-AC433US
Device Descriptor:
 bLength                18
 bDescriptorType         1
 bcdUSB               2.01
 bDeviceClass            0 
 bDeviceSubClass         0 
 bDeviceProtocol         0 
 bMaxPacketSize0        64
 idVendor           0x04bb I-O Data Device, Inc.
 idProduct          0x0955 
 bcdDevice            1.00
 iManufacturer           1 I-O DATA
 iProduct                2 WHG-AC433US
 iSerial                 3 3476C5850288
 bNumConfigurations      1
 Configuration Descriptor:
   bLength                 9
   bDescriptorType         2
   wTotalLength       0x004a
   bNumInterfaces          1
   bConfigurationValue     1
   iConfiguration          0 
   bmAttributes         0xa0
     (Bus Powered)
     Remote Wakeup
   MaxPower              160mA
   Interface Descriptor:
     bLength                 9
     bDescriptorType         4
     bInterfaceNumber        0
     bAlternateSetting       0
     bNumEndpoints           8
     bInterfaceClass       255 Vendor Specific Class
     bInterfaceSubClass      2 
     bInterfaceProtocol    255 
     iInterface              0 
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x84  EP 4 IN
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x85  EP 5 IN
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x08  EP 8 OUT
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x04  EP 4 OUT
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x05  EP 5 OUT
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x06  EP 6 OUT
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x07  EP 7 OUT
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
     Endpoint Descriptor:
       bLength                 7
       bDescriptorType         5
       bEndpointAddress     0x09  EP 9 OUT
       bmAttributes            2
         Transfer Type            Bulk
         Synch Type               None
         Usage Type               Data
       wMaxPacketSize     0x0200  1x 512 bytes
       bInterval               1
Binary Object Store Descriptor:
 bLength                 5
 bDescriptorType        15
 wTotalLength       0x000c
 bNumDeviceCaps          1
 USB 2.0 Extension Device Capability:
   bLength                 7
   bDescriptorType        16
   bDevCapabilityType      2
   bmAttributes   0x00000002
     HIRD Link Power Management (LPM) Supported
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
 (Bus Powered)
```

morrownr, 2023-03-01:
I think we solved the issue so I am closing this for now. If you have additional issues, please post a new issue. Thanks.