morrownr/88x2bu-20210702 #119: issue during the utilization on RPI 3 arm64

mattex121, 2023-01-21:
Hello,
I start by thanking you for the drivers and for the work.
I proceeded with the installation following the detailed instructions in the readme file, the installation takes place without any problems.
Once the raspberry restarted I checked that the adapter was among the devices connected via usb with "lsusb" and it is detected but then it doesn't belong in the network interfaces.
How can I fix?
I currently have the following kernel: 5.15.84-v8+
it's a fresh install of the entire OS and I'm trying to use an "ALFA AWUS036ACH" as a network adapter.
I also state that I have no interest in using the monitor mode of the adapter but I only want to use it to connect to an existing wifi.

Thank you in advance

morrownr, 2023-01-22:
Hi @mattex121 

> "ALFA AWUS036ACH"

The Alfa ACH uses a rtl8812au chipset so you have installed the wrong driver.

The site Main Menu is located:

https://github.com/morrownr/USB-WiFi

Select menu item 5 for Realtek info. The section on the rtl8812au will have a link to that driver.

Before installing the correct driver, run `sudo ./remove-driver.sh` and then delete the 88x2bu directory.

Good luck. The Alfa ACH is a nice adapter with great range. I have one. It does use a lot of power so be careful about how many items you have in the usb ports.

Nick

mattex121, 2023-01-23:
What a stupid mistake i made, among the thousands of guides I've read I must have taken the wrong link, I'm really sorry I wasted your time.

Thank you very much for your availability,
Matteo

closed 2023-01-23 by mattex121

morrownr, 2023-01-23:
You are welcome and I hope you enjoy the adapter.

Nick