morrownr/88x2bu-20210702 #120: AP not working with Comfast CF-924AC V2

DodoLeDev, 2023-01-21:
### Hi everyone!
I've bought a Wi-Fi adapter supposed to work with Linux out-of-the-box, but I figured out that I needed this driver to make it work (so yes, it works with my model).

Unfortunately, I can't get the AP to work... Enabling the option ends up with a timeout from the _Authorization Applicant_:
[image]
(Translation: the time limit given to the Authorization Applicant is exceeded)

After looking into the kernel logs, I faced with these two errors/warnings:
[image]
[image]

---

## System information:
**Wifi adapter**: Comfast CF-924AC V2
**OS:** I'm using Arch Linux with KDE 5.26.5 on Wayland

| uname -mr | lsusb | dkms status | rfkill list all |
| --- | --- | --- | --- |
| `6.1.7-arch1-1 x86_64` | `Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]` | `rtl88x2bu/5.13.1, 6.1.7-arch1-1, x86_64: installed` | `0: phy0: Wireless LAN         Soft blocked: no        Hard blocked: no` |

### `iw dev` command:
```phy#0
        Interface wlp0s20f0u3
                ifindex 2
                wdev 0x1
                addr 20:0d:b0:c5:a4:53
                ssid Bbox-490FD16B
                type managed
                channel 100 (5500 MHz), width: 40 MHz, center1: 5510 MHz
                txpower 15.00 dBm
```

### `iw reg get` command:
```global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

*Any help is welcome!*

morrownr, 2023-01-22:
Hi @DodoLeDev 

What computer are you using?

What software are you using for ap support?

Main Menu item 9 gives extensive information about setting up an ap:

https://github.com/morrownr/USB-WiFi

DodoLeDev, 2023-03-01:
Hi! Sorry for the – extremely – late reply, but your help in the _Main menu item 9_ solved my issue! 😅

Thank you for all!

morrownr, 2023-03-01:
Oh, this is funny. There is late and then there is really late.

Glad you got it going.