morrownr/88x2bu-20210702 #122: How to remove old driver files? - /var/lib/dkms/rtl88x2bu/5.8.7.4/source/dkms.conf does not exist

ThiloteE, 2023-02-04:
Sorry, I am the user from very old [issue #54](https://github.com/morrownr/88x2bu/issues/54) . Not knowing it any better, I messed up updating and updated my Fedora system multiple times to newest kernel without prior removal of the 88x2bu driver... Sorry, I have learned my lesson. Will not happen again, but how can I now remove old stuff?

Found following error:

```
[Thilo@server 88x2bu-20210702]$ dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl88x2bu/5.8.7.4/source/dkms.conf does not exist.
```

In short what I did:

1. Installed old version of rtl88x2bu driver more than a year ago.
2. Forgot I had the driver still installed, because I started to use cables on this machine.
2. Updated Fedora multiple times, including kernel. Currently at kernel 6.1, if i remember correctly.
3. Found dkms error messages
4. Went to 88x2bu directory and used `sudo ./remove-driver.sh`; Rebooted
5. dkms error messages remained
6. Cloned newes version of 88x2bu-20210702 from 2023-02-04
7. Went into 88x2bu-20210702 directory and used `sudo ./remove-driver.sh`; Rebooted
8. dkms error messages remained
9. Went into 88x2bu-20210702 directory and used `./install-driver.sh`; Rebooted
10. Went into 88x2bu-20210702 directory and used `sudo ./remove-driver.sh`; Rebooted
11. dkms error messages remained

```
[Thilo@server src]$ cd 88x2bu
[Thilo@server 88x2bu]$ ls
README.md
[Thilo@server 88x2bu]$ cd ..
[Thilo@server src]$ cd 88x2bu-20210702
[Thilo@server 88x2bu-20210702]$ ls
88x2bu.conf  LICENSE   README.md  default-editor.txt  dkms.conf  edit-options.sh  halmac.mk  install-driver.sh  platform          rtl8822b.mk  supported-device-IDs
Kconfig      Makefile  core       dkms-make.sh        docs       hal              include    os_dep             remove-driver.sh  save-log.sh
[Thilo@server 88x2bu-20210702]$ dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl88x2bu/5.8.7.4/source/dkms.conf does not exist.
[Thilo@server 88x2bu-20210702]$ cd /var/lib/dkms/
[Thilo@server dkms]$ ls
mok.key  mok.pub  rtl88x2bu
[Thilo@server dkms]$ cd rtl88x2bu
[Thilo@server rtl88x2bu]$ ls
5.8.7.4                                kernel-5.13.15-200.fc34.x86_64-x86_64  kernel-5.14.14-200.fc34.x86_64-x86_64  kernel-5.14.16-201.fc34.x86_64-x86_64  kernel-5.14.9-200.fc34.x86_64-x86_64
kernel-5.13.12-200.fc34.x86_64-x86_64  kernel-5.14.10-200.fc34.x86_64-x86_64  kernel-5.14.15-200.fc34.x86_64-x86_64  kernel-5.14.17-201.fc34.x86_64-x86_64
[Thilo@server rtl88x2bu]$ 
```

ThiloteE, 2023-02-04:
Found related issue: https://unix.stackexchange.com/questions/416912/why-can-i-not-remove-this-dkms-module

ThiloteE, 2023-02-04:
Closing this issue. Not completely sure, but I think following advise solved it.

> 
> 
> In case of normal operations gone wrong, you can always delete dkms add-ons by hand, with sudo or as root.
> 
> Normally the modules sources are installed by make install under /var/lib/dkms/ in a directory with the corresponding name, probably named rtl...something. Just delete that directory.
> 
> You have also to delete the corresponding compiled file module under /lib/modules/KERNEL_VERSION/updates/dkms/ where KERNEL_VERSION is your current kernel. The file should be called rtl...something.ko or similar.
> 
> Once done that, you can either try to rmmod the module or falling that, reboot.
> 
> No module and corresponding dkms anymore in the system.
> 
> While not critical, the module dependencies also need to be updated after deleting the module. Run:
> 
> sudo /sbin/depmod -a 
> 

> 
> [edited Feb 17, 2018 at 18:38](https://unix.stackexchange.com/posts/416913/revisions)
> answered Jan 13, 2018 at 21:33
> [Rui F Ribeiro](https://unix.stackexchange.com/users/138261/rui-f-ribeiro)'s user avatar
> Rui F Ribeiro
> 54.8k2626 gold badges144144 silver badges221
> 

Using `dkms status` yields no error message anymore.

Looking forward to in-kernel drivers for rtl88x2bu 👍 Also, thank you Nick for your work here. I have been recommending your repository for a while now :-)

closed 2023-02-04 by ThiloteE

morrownr, 2023-02-05:
Hi @ThiloteE 

> Looking forward to in-kernel drivers for rtl88x2bu +1 Also, thank you Nick for your work here. I have been recommending your repository for a while now :-)

The work I have been merging over the last few weeks is partly aimed at eliminating issues such as you experienced. The bottom line is that these Realtek drivers do not age well which is why I encourage users to go with adapters that are supported with in-kernel drivers. I hope my recent additions help but something to remember is that anytime you are doing an upgrade that includes more than a kernel point upgrade (5.15.4 > 5.15.7) you really need to upgrade your driver first via...

$ git pull
$ sudo ,/install-driver.sh

The updated docs do included secure boot instructions for fedora now. If you have any more problems, let me know.

Nick