morrownr/88x2bu-20210702 #125: Slight warning after upgrading to latest kernel on ubuntu 22.04

altineller, 2023-02-21:
Hello,

During boot, I get the following warning message:

UBSAN: invalid-load in /var/lib/dkms/rt188x2bu/5.13.1/build/hal/phydm/phydm_ccx.c
load of value 255 is not a valid value for type '_Bool'

What changes can we do in the code to fix it?

Best Regards,
C.A.

morrownr, 2023-02-21:
Hi @altineller 

> hal/phydm/phydm_ccx.c

I'm looking but that is a big file. Does this show in dmesg also? If so, does it show a line number?

I'm pretty sure this can be cleaned up quickly if I can just get on the code that is causing the problem.

altineller, 2023-02-22:
Hello @morrownr

Below are relevant parts of dmesg. I believe the line number is given here:

UBSAN: invalid-load in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_ccx.c:694:2

Best Regards,
Can

[    9.756247] RTW: hal_com_config_channel_plan country code:"TR" with chplan:0x26
[    9.781987] audit: type=1400 audit(1662749298.603:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="lsb_release" pid=579 comm="apparmor_parser"
[    9.789484] audit: type=1400 audit(1662749298.611:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=580 comm="apparmor_parser"
[    9.789521] audit: type=1400 audit(1662749298.611:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=580 comm="apparmor_parser"
[    9.805214] audit: type=1400 audit(1662749298.627:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/man" pid=582 comm="apparmor_parser"
[    9.805244] audit: type=1400 audit(1662749298.627:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_filter" pid=582 comm="apparmor_parser"
[    9.805258] audit: type=1400 audit(1662749298.627:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_groff" pid=582 comm="apparmor_parser"
[    9.832517] audit: type=1400 audit(1662749298.655:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=581 comm="apparmor_parser"
[    9.833074] audit: type=1400 audit(1662749298.655:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-helper" pid=581 comm="apparmor_parser"
[    9.833356] audit: type=1400 audit(1662749298.655:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/connman/scripts/dhclient-script" pid=581 comm="apparmor_parser"
[    9.833376] audit: type=1400 audit(1662749298.655:11): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/{,usr/}sbin/dhclient" pid=581 comm="apparmor_parser"
[    9.871979] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R
[    9.871996] RTW: [RF_PATH] HALSPEC's rf_reg_trx_path_bmp:0x33, rf_reg_path_avail_num:2, max_tx_cnt:2
[    9.872003] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[    9.872007] RTW: [RF_PATH] Registry's trx_path_bmp:0x00, tx_path_lmt:0, rx_path_lmt:0
[    9.872012] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[    9.872017] RTW: [RF_PATH] HALDATA's rf_type:RF_2T2R, NumTotalRFPath:2
[    9.872023] RTW: [TRX_Nss] HALSPEC - tx_nss:2, rx_nss:2
[    9.872028] RTW: [TRX_Nss] Registry - tx_nss:0, rx_nss:0
[    9.872032] RTW: [TRX_Nss] HALDATA - tx_nss:2, rx_nss:2
[    9.872039] RTW: txpath=0x3, rxpath=0x3
[    9.872043] RTW: txpath_1ss:0x1, num:1
[    9.872048] RTW: txpath_2ss:0x3, num:2
[    9.872870] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[    9.874835] RTW: rtw_ndev_init(wlan0) if1 mac_addr=74:da:38:dc:6b:89
[    9.875764] usbcore: registered new interface driver rtl88x2bu
[    9.875783] RTW: module init ret=0
[    9.911050] rtl88x2bu 1-1.2:1.0 wlx74da38dc6b89: renamed from wlan0
[   11.399969] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[   11.423496] IPv6: ADDRCONF(NETDEV_CHANGE): wlx74da38dc6b89: link becomes ready
[   11.561466] ================================================================================
[   11.561502] UBSAN: invalid-load in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_ccx.c:694:2
[   11.561517] load of value 255 is not a valid value for type '_Bool'
[   11.561530] CPU: 2 PID: 614 Comm: RTW_CMD_THREAD Tainted: G         C OE     5.15.0-1024-raspi #26-Ubuntu
[   11.561538] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   11.561542] Call trace:
[   11.561545]  dump_backtrace+0x0/0x1f0
[   11.561557]  show_stack+0x20/0x30
[   11.561562]  dump_stack_lvl+0x8c/0xb8
[   11.561570]  dump_stack+0x18/0x34
[   11.561574]  ubsan_epilogue+0x10/0x54
[   11.561579]  __ubsan_handle_load_invalid_value+0x80/0x90
[   11.561587]  phydm_nhm_mntr_set+0x168/0x16c [88x2bu]
[   11.561867]  phydm_env_mntr_trigger+0x54/0x174 [88x2bu]
[   11.562034]  rtw_acs_trigger+0x98/0x120 [88x2bu]
[   11.562200]  sitesurvey_cmd_hdl+0xdd4/0x1048 [88x2bu]
[   11.562366]  rtw_cmd_thread+0x448/0x768 [88x2bu]
[   11.562531]  kthread+0x128/0x13c
[   11.562539]  ret_from_fork+0x10/0x20

morrownr, 2023-02-22:
Can,

Yes, thank you. I'll take a look as soon as I can.

@morrownr

morrownr, 2023-03-01:
@altineller 

I've been looking at this but so far I have not been able to determine exactly what is going on or why. I'll keep an eye on this. If you run across anything that would help, let me know.

@morrownr