morrownr/88x2bu-20210702 #127: Bad range performance when using TP-Link T4U v.3 vs cheaper interfaces

nanoR4K, 2023-02-28:
[image]

Reference:

Left: T2U plus rtl8811au with rtl88xxau aircrack-ng drivers
Right: T4U v.3 rtl8812bu with the latest release of your driver

OS: Kali Linux freshly updated

morrownr, 2023-03-01:
Hi @nanoR4K 

I don't have an explanation for what you are seeing. Range can be affected by various things. This repo averages 305 clones per day and I am not seeing much being reported about range. My own testing shows good results:

iperf3 test performance a few minutes ago:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.223 port 55274 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  60.8 MBytes   510 Mbits/sec    0   1.81 MBytes       
[  5]   1.00-2.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   2.00-3.00   sec  63.8 MBytes   535 Mbits/sec    0   1.81 MBytes       
[  5]   3.00-4.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   4.00-5.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   5.00-6.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   6.00-7.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   7.00-8.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   8.00-9.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   9.00-10.00  sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   645 MBytes   541 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   642 MBytes   538 Mbits/sec                  receiver

```

My point is that you can't get results like this without a good signal and there is about 20 feet and 2 walls between the adapter and AP in this case so range is not causing problems.

Here is an injection test:

```
$ sudo aireplay-ng --test wlx12121212
[sudo] password for me:         
23:28:25  Trying broadcast probe requests...
23:28:25  Injection is working!
23:28:27  Found 50 APs

23:28:27  Trying directed probe requests...
23:28:27  A0:FF:70:F7:D3:A5 - channel: 157 - ''
23:28:28  Ping (min/avg/max): 0.292ms/1.512ms/5.352ms Power: -36.75
23:28:28  28/30:  93%

23:28:28  A0:FF:70:F7:D3:A9 - channel: 157 - ''
23:28:28  Ping (min/avg/max): 0.292ms/1.594ms/2.422ms Power: -36.93
23:28:28  29/30:  96%

...
```

Range is not too bad. It found 50 APs and was injecting at a high rate.

There is a rtl8821au (for the 8811cu adapter) driver here as well if you want to test it against the aircrack driver.

nanoR4K, 2023-03-01:
Maybe this particular to link adapter is defective ? Because I tested it on a few different scenarios and it always get either a weaker or no signal while the other two (both tp link T2U and the plus version) can get a lot more.

nanoR4K, 2023-03-01:
Do I need to run `airmon-ng check kill && airmon-ng start wlan1` to perform both of this tests ? Im replicating them so you can have a better ideia

morrownr, 2023-03-01:
> Do I need to run airmon-ng check kill && airmon-ng start wlan1 to perform both of this tests ?

The injection test should have interfering processes paused or killed before running the test but let me recommend an alternative to airmon-ng:

Go to the site Main Menu and select item 10:

https://github.com/morrownr/USB-WiFi

I use `start-mon.sh` instead of airmon.ng as, in my opinion, it does a better job. You can read about it in the docs.

Then you can run the following to test injection:

sudo aireplay-ng --test <wlan0>

Regarding iperf3. iperf3 is used to test LAN throughput. You need an iperf3 server somewhere on your LAN. I run OpenWRT on my main WiFi router and it has an iperf3 package so I run iperf3 on my router. You could also connect a RasPi4B to your router via cat 5+ cable. The iperf3 could run on the Pi. You may want to do a seach for articles about setting up iperf3 so as to get other options. iperf3 really can give you info about connections that can help solve issues.

nanoR4K, 2023-03-01:
After running the injection test it gets around 2-4 APs while the other one 10-15 on the same area. Another detail I noticed is the txpower when running `iw dev`, that being 16 db to 18db max.

Besides that Injection is working fine with all 3 found networks getting 100% scores at 60% to 30% power

morrownr, 2023-03-01:
> After running the injection test it gets around 2-4 APs while the other one 10-15 on the same area.

I have a lot of usb wifi adapters and I am very aware that there is a lot of variability in range performance. Menu item 2 on the Main Menu link I posted above shows the recommended adapters that I and others maintain. In that list, there is one adapter that smokes the others. It is the Alfa ACHM. It is a little single antenna, AC600 adapter but the range on that little adapter simply amazing.

Range mostly has to do with the amps and antennas that the adapter makers use. Drivers really have very little to do with range. I hear that Alfa will be bringing a new "Max Power" adapter nid year and it will be based on the mt7921au WiFi 6e chipset that already has a stable driver in the kernel. Alfa calls their long range adapter "Max Power" but power really has little to do with it. They simply use very very high quality amps and antennas to get that long range. Those adapters are not cheap but they are good.

closed 2023-03-01 by nanoR4K