morrownr/88x2bu-20210702 #129: Have to remove and install driver again after lost power (Ubuntu)

xhuan8, 2023-03-06:
If my Ubuntu server lost power and shutdown unexpectly, the driver won't work on the next boot,

have to perform remove-reboot-install-reboot to make the wifi icon show up,

any advice to make the recovery easier?

morrownr, 2023-03-06:
Try this:

$ git pull
$ sudo ./install-driver.sh

morrownr, 2023-03-06:
Try this:

sudo remove -m rtl88x2bu -v 5.8.7.4 --all

Then remove and reinstall your header files.

Then:

$ sudo ./install-driver.sh

xhuan8, 2023-03-07:
Do you mean "sudo apt-get remove"?
found nothing

morrownr, 2023-03-07:
Opps... typo. That first command should be:

sudo dkms remove -m rtl88x2bu -v 5.8.7.4 --all

That should remove all of the below:

```
Checking for previously installed drivers.
rtl88x2bu, 5.8.7.4, 5.4.0-128-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-131-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-132-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-135-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-136-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-137-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-139-generic, x86_64: installed
rtl88x2bu, 5.8.7.4, 5.4.0-144-generic, x86_64: installed
```

We need to uninstall/remove the older version of the driver. I made a note in the source for my scripts to write code to account for this when able but for now, it is detecting the issue but is not yet capable of resolving it for a driver this old so we need to do it manually. Once that is complete, follow the rest of the instructions about so that we can see where we are.

xhuan8, 2023-04-26:
Still not working

: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 56/56 (in-use/total processing units)
: 65816848 (total system memory)
: 5.15.0-69-generic (kernel version)
: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
: dkms:2.8
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
                 /usr/src/rtl88x2bu-5.13.1

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh...(bad exit status: 126)
ERROR (dkms apport): binary package for rtl88x2bu: 5.13.1 not found
Error! Bad return status for module build on kernel: 5.15.0-69-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 0.55 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2023-04-26:
This may be a header files problem. I don't have an installation of Ubuntu server to test. You could try this:

```
$ sudo apt remove build-essential
```

```
$ sudo apt install build-essential
```

xhuan8, 2023-04-26:
the version 20221007 works good, I'm not sure what's the difference.
maybe I'll try the latest version later.

morrownr, 2023-04-26:
@xhuan8 

I'm going to make a suggestion. Servers generally need to be very stable with high uptime. Why not use a wireless adapter that is very stable and has high uptime? 

Here is the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

Menu item 2 has numerous adapters that use in-kernel drivers. In-kernel drivers are inherently more stable and trouble free. If you pass on your requirements for range, throughput and thermal characteristics, I can make a recommendation or two.

xhuan8, 2023-04-26:
You are right, but currently I'll leave it as is.

closed 2023-04-26 by xhuan8