morrownr/88x2bu-20210702 #130: warning: the compiler differs from the one used to build the kernel

flyingpreacher, 2023-03-06:
I am on OpenMandriva Lx release 23.03 (ROME) Rolling for x86_64
Here is what I have installed so far: 
clang
dkms
git
kernel-desktop-devel
kernel-headers
task-devel
bc

I started the installation with ```sudo ./install-driver.sh```
Here is what I have as my output. Thanks in advance for your help!

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.2-desktop-1omv2390/build M=/home/miltonsj/src/88x2bu-20210702  modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: OpenMandriva 15.0.7-3 clang version 15.0.7 (/builddir/build/BUILD/llvm-project-15.0.7.src/clang c5b761db9d28a8cbbafbaa455d57066c5a0e42bc)
  You are using:           gcc (GCC) 12.2.0 20220819 (OpenMandriva)
  CC [M]  /home/miltonsj/src/88x2bu-20210702/core/rtw_cmd.o
  CC [M]  /home/miltonsj/src/88x2bu-20210702/core/rtw_security.o
  CC [M]  /home/miltonsj/src/88x2bu-20210702/core/rtw_debug.o
  CC [M]  /home/miltonsj/src/88x2bu-20210702/core/rtw_io.o
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /home/miltonsj/src/88x2bu-20210702/core/rtw_security.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /home/miltonsj/src/88x2bu-20210702/core/rtw_debug.o] Error 1
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /home/miltonsj/src/88x2bu-20210702/core/rtw_cmd.o] Error 1
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /home/miltonsj/src/88x2bu-20210702/core/rtw_io.o] Error 1
make[1]: *** [Makefile:2021: /home/miltonsj/src/88x2bu-20210702] Error 2
make: *** [Makefile:2506: modules] Error 2
An error occurred:  2

morrownr, 2023-03-06:
> clang

From the README:

gcc 10, 11 and 12

I don't test with clang due to the fact that there are not enough hours in the day for me to accomplish the priority issues around here. If you discover how to make clang work, I'll be glad to add it to the documentation.

With that said, recommend you install gcc and give it a try.

@morrownr

flyingpreacher, 2023-03-06:
Thanks for your answer. I will check and see what I can do.