morrownr/88x2bu-20210702 #132: dkms compile error on ORACLE LINUX 8.7

jtome7, 2023-03-13:
I have successfully installed this before on ORACLE LINUX 8 on another machine, so I might be missing a pkg or something on this machine??

/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:470:3: error: too many arguments to function ‘cfg80211_ch_switch_started_notify’
   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7031:6: note: declared here
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:8398:9: warning: ‘struct mgmt_frame_regs’ declared inside parameter list will not be visible outside of this definition or declaration
  struct mgmt_frame_regs *upd)
         ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_update_mgmt_frame_register’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:8426:8: error: dereferencing pointer to incomplete type ‘struct mgmt_frame_regs’
    (upd->interface_stypes & rtw_stypes_mask)) {
        ^~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10521:3: error: ‘struct cfg80211_ops’ has no member named ‘update_mgmt_frame_registrations’; did you mean ‘mgmt_frame_register’?
  .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   mgmt_frame_register
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10521:37: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, u64)’ {aka ‘int (*)(struct wiphy *, struct wireless_dev *, long long unsigned int)’} from incompatible pointer type ‘void (*)(struct wiphy *, struct wireless_dev *, struct mgmt_frame_regs *)’ [-Werror=incompatible-pointer-types]
  .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10521:37: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1798: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.4.17-2136.316.7.el8uek.x86_64'
make: *** [Makefile:2506: modules] Error 2

morrownr, 2023-03-14:
Hi @jtome7 

Is Oracle Linux 8.7 based on RHEL?

jtome7, 2023-03-14:
Hello there, Nick!   Hope all is well.

Yes ... Oracle Linux is that same general family as RHEL, CentOS, Fedora
and the newer distros Rocky, Alma and ClearOS.

What's curious about my attempt to install the driver on a new PC is that I
have installed it successfully several times before on different hardware
and still have it running there.  So, I'm wondering if this new machine I'm
setting up is missing a package or what would be different?

As always ... THANKS for your help.
*RedHat-based Linux Distributions*
https://www.tecmint.com/redhat-based-linux-distributions/

morrownr, 2023-03-15:
> Hello there, Nick!   Hope all is well.

I'm a little under the weather but am trying to get a few things done. I hope all is well with you.

> Yes ... Oracle Linux is that same general family as RHEL, CentOS, Fedora
and the newer distros Rocky, Alma and ClearOS.

Hmmm... general family. Well, from the perspective of this driver, there are standard kernels, such as is used in fedora, that work well. Then there are distros such as RHEL that do some things to the kernel they use that makes it very difficult for me to support. In fact, I don't support RHEL or distros that use RHEL kernels. It would take too much time. I do offer to allow RHEL devs to merge the capability for this driver to work on specific releases of RHEL. Currently 2 kernels are supported.

The log you posted tells me you are trying to install on a RHEL kernel that does not yet have support. If your use case allows, you might want to look at using fedora.

morrownr, 2023-03-15:
Here is the msg that RHEL dev left when first submitted the last patch:

```
Question: Were compromises made to make this work on EL8 (RHEL, CentOS, Rocky, etc) distributions?

Answer: Yes. EL8 distribution kernels contain many backports of features and fixes from newer kernels, but sometimes in ways that break drivers that are not specifically tested against. For EL 8.6 specifically, the driver contains a significant hack that ignores the current link state of the adapter when transmitting or receiving frames. This matches the behavior of ancient drivers for other chipsets that do work on EL 8.6 (as they do not check link status at all), but is really a kludge to work around a station accounting issue on these kernels. Only Ad-Hoc and Station mode have been tested.
```

I have to keep an eye on the RHEL patches so that I can rapidly pull them if a problem comes up. I can go into detail about problems RHEL causes and even show you some patches in the source if you like.

closed 2023-03-24 by morrownr