morrownr/88x2bu-20210702 #133: Adapter disconnects despite having rtw_power_mgnt=0

fs30000, 2023-03-14:
Hello guys, so i'm using a TPLink T3U Plus in AP mode. Even tough the module is configured with rtw_power_mgnt=0, it disconects.

```Mar 13 02:33:27 cube kernel: usb 2-3: USB disconnect, device number 3
Mar 13 02:33:27 cube kernel: br0: port 2(wlp0s20f0u3) entered disabled state
Mar 13 02:33:27 cube kernel: device wlp0s20f0u3 left promiscuous mode
Mar 13 02:33:27 cube kernel: br0: port 2(wlp0s20f0u3) entered disabled state
Mar 13 02:33:27 cube kernel: RTW: rtw_ndev_uninit(wlp0s20f0u3) if1
Mar 13 02:33:27 cube NetworkManager[831]: <info>  [1678674807.5599] device (wlp0s20f0u3): state change: activated -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Mar 13 02:33:27 cube NetworkManager[831]: <info>  [1678674807.5603] device (br0): bridge port wlp0s20f0u3 was detached
Mar 13 02:33:27 cube NetworkManager[831]: <info>  [1678674807.5604] device (wlp0s20f0u3): released from master device br0```

As you can see, it is bridged to br0. Anything, i can do to avoid this?

morrownr, 2023-03-14:
Hi @fs30000 

Power saving is very complicated in this driver, The default `rtw_power_mgnt=1` has proven to be very dependable and I have found no advantage in using the other settings in most cases. With that said, setting  `rtw_power_mgnt=0` actually does not fully shut off power saving.

To fully shut down power saving, you need to remove any entries in 88x2bu.conf that are power saving related (and there are many) and fine the following line in `Makefile`:

CONFIG_POWER_SAVING = y

change it to:

CONFIG_POWER_SAVING = n

Save and reinstall with

$ sudo sh install-driver.sh

Don't worry about removing the driver first as the `install-driver.sh` script has the capability these days to detect and remove drivers that it previously installed.

fs30000, 2023-03-15:
Ok, recompiled it. Hopefully i'll remember to report back after some days/weeks :)
Thanks for the tip.