morrownr/88x2bu-20210702 #134: Log spam with start_addr, buffer_size

alkisg, 2023-03-15:
After updating to the latest driver version, my dmesg is spammed with these messages:

```
[Μαρ15 09:18] usbcore: registered new interface driver rtl88x2bu
[  +0,029266] rtl88x2bu 3-10:1.0 wlx4401bb912885: renamed from wlan0
[  +0,699922] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[  +5,189512] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[Μαρ15 09:19] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[  +8,996556] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[ +12,999659] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
```

I tried `git bisect` and it said that commit fb6b4cda6a30aa7e3fc07d1f92acdeabcaae6e26 is to blame:

```
commit fb6b4cda6a30aa7e3fc07d1f92acdeabcaae6e26
Author: morrownr
Date:   Mon Jan 9 18:16:40 2023 -0600

    significant upgrade - (1)
```

I'll see if I can pinpoint the exact line that caused the error; any help appreciated.

alkisg, 2023-03-15:
I think it's caused by this changed line in phydm_debug.h:

```diff
- #define pr_debug		_RTW_DBG
+ #define pr_debug		printk
```

To me it looks like this change forces everything to go to dmesg, instead of going through the debug functions; thus disabling all log levels. Should we revert that?

morrownr, 2023-03-17:
Reverted. Let me know if the problem is fixed.

alkisg, 2023-03-17:
Hi Nick, yes I've tested the reversal yesterday and it fixed the issue in all drivers, not in just this one. Thanks!

closed 2023-03-17 by alkisg