morrownr/88x2bu-20210702 #136: Install to Raspberry PI4 latest kernel 6.1.19 on bullseye

Windysurf, 2023-03-22:
Hi,

I try to install this driver on the kernel 6.1.19.
It is no problem and works fine on the kernel 5.15.
Recentry I update the kernel from 5.15. to 6.1.

USB-wifi : tp-link Archer T3U Plus

After typing ./install-driver.sh, the error message is as follows,

"Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh" "

Thank you for your help

morrownr, 2023-03-22:
Hi @Windysurf 

Try:

```
sudo apt install -y raspberrypi-kernel-headers
```

Windysurf, 2023-03-22:
I tried this, but it shows the version is already newest (1:1.20230317-1).

Thanks

morrownr, 2023-03-22:
>  tried this, but it shows the version is already newest

The newest is not what matters with header files. Each kernel has header files that match it. Kernel 6.1.19 requires the 6.1.19 header files.

The Raspberry Pi OS is known to have problems along these lines. The RasPi devs really need to fix this because this is their problem, not a problem with this driver. Please report this to Raspberry Pi.

Try:

```
sudo apt remove -y raspberrypi-kernel-headers
```

```
sudo apt install -y raspberrypi-kernel-headers
```

An alternative is to get an adapter that uses in-kernel drivers. See menu item 2 at the site Main Menu:

https://github.com/morrownr/USB-WiFi/blob/main/README.md

Windysurf, 2023-03-22:
Thank you for your reply.

I tried remove and install the headers but not solved.

I understood the situation.
I still use the drivers on the kernel 5.15 until the rt8812bu in-kernel driver will be supported / fix the problem, or change the adapter.

Windysurf, 2023-03-22:
Hi,

I solved this issue.
I can make file and connect to wifi succesfully.

refer to :
[https://forums.raspberrypi.com/viewtopic.php?p=2091532&hilit=Tp+link#p2091532](url)

32bit  OS and 64bit kernel mismatch.

I add  " arm_64bit=0" in the /boot/config.txt and after reboot, ./install-driver.sh. 
The Pi 4 firmware now prefers the 64-bit kernel if one exists. Setting arm_64bit=0 forces it to use the 32-bit kernel.

Thanks

morrownr, 2023-03-22:
Bad decision by the RasPi engineers. They are breaking userspace by creating a mismatch.

morrownr, 2023-03-22:
@Windysurf 

FYI: I just tested. You can avoid this issue by using the 2023-02-21 64 bit RasPiOS.

https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit

Prior versions had header issues but that is cleaned up now so I guess they had to create more header issues somewhere else but a clean install of the above followed by an upgrade should work well if you are inclined to do so.

Windysurf, 2023-03-22:
Thank you for your advice.