morrownr/88x2bu-20210702 #138: Driver doesn't auto-install after kernel update

viggen66, 2023-03-27:
On previous driver version, dated july 2022 and before, whenever a new kernel update came out, the Wi-FI driver always installed along with DKMS, but on recent versions it doesn't, it requires the run the script install-driver.sh, is there a way to make it auto-install as before?

morrownr, 2023-03-29:
Hi @viggen66 

Did you run the `remove-driver.sh` script with the old version of the driver before cloning and installing the new version?

viggen66, 2023-03-29:
Hi morrownr,

To begin with, thanks for your effort and support, your driver really works well with Linux, I see it this way, for some ubuntu distros' like Zorin, Mint, the older version (July 2022) restores Wifi driver on Kernel updates, but for newer kernel versions namely 5.19.0-38-generic, the older version driver doesn't work, only the most recent one of 2023, also KDE did a kernel update and Wifi was working, it seems there's some version of Linux kernel that makes the jump between 2022 and 2023.