morrownr/88x2bu-20210702 #139: wifi randomly disconnects TP-LINK ac1300 Archer T4U Plus

jurassicjordan, 2023-04-03:
since updating to the latest Arch kernel (6.2.9-arch1-1) the network connection on my home recycled server box has been randomly disconnecting, closing connection with the network and showing no available connections without removing the wifi adapter and turning off and then back on the wifi connection and plugging it back in (not always in that order) or the wifi connection will break, showing a connection in but the wifi symbol (in KDE anyway) turns blue. 

6.2.9-arch1-1
EndeavourOS Linux x86_64 
Plasma 5.27.3 

Let me know if there is any additional info i can provide.

jurassicjordan, 2023-04-03:
the light on the adaptor used to blink before, just thought id mention that

julipetric, 2024-08-06:
After some digging, I ended up with this thread. The driver update from the first message (original question edit) solved it for me. It even revived the green status led
https://forums.linuxmint.com/viewtopic.php?t=398504

morrownr, 2024-08-06:
@julipetric 

Sorry for being slow to help. I've been busy.

The thread that you posted from Linux Mint contains a LOT of bad advice. If you want me to go over things with you, I will.

julipetric, 2024-08-06:
Hi, if you could explain further why it isn't proper that would be great.
By the way, the code I was referring to was this bit

```sudo dkms remove rtl88x2bu/5.8.7.1 --all
sudo apt install git dkms
git clone https://github.com/morrownr/88x2bu-20210702.git
cd 88x2bu-20210702
sudo ./install-driver.sh
```

As far as I understood, it removes any instance of the driver, clones the repo and installs it again from the contained script.

Thanks for the reply, and no need to apologize :)

morrownr, 2024-08-06:
>  if you could explain further why it isn't proper that would be great.

Sure.

One of the first suggestions is to turn power save off. Well, it has been years since I have seen that fix anything, Power save is not really problematic in this day in time. 10-12 years ago power save would cause some problems with some drivers.

I also saw the below:

> Can you remove those old drivers in Synaptic? Only one you should keep is the 6.2.0 one:

Synaptic can only remove a driver if it is installed as a deb or by Synaptic. You can tell this by looking at what they were looking at.

What was the point in keeping the 6.2.0 one? 

What we were seeing is normal. As new kernels are installed, dkms will compile a new driver. The old one were all in a single kernel so there was no conflict. That was not the problem.

> git clone https://github.com/cilynx/rtl88x2bu.git

They then sent the user to unstall an old driver that is likely not aware that kernel 6.2 added an internal driver which will conflict.

I could go on but the advice in that thread was not good. I like that they tried. They just did not know.

The driver in this repo has considered and taken care of the conflict with `install-driver.sh` The script blacklist the internal driver so that there is no conflict. 

> As far as I understood, it removes any instance of the driver,

What you did should have worked but it cannot detect all possible conflicts because there are distros and maintainers of repos that rename or otherwise that things they really should not do which makes it impossible for me to write code to catch everything.

Are you seeing any current problems?