morrownr/88x2bu-20210702 #140: Radiotap in injecting frame

alza36sc105, 2023-04-05:
Can the driver support interpreting the Radiotap in injecting frame?
I want to inject frames at various transfer rates, but rtw_monitor_xmit_entry seems to be ignoring Radiotap's interpretation.
Therefore, it is always transferred at 1mbps when injecting...

morrownr, 2023-04-06:
Hi @alza36sc105 

It has been a while since I looked at that code. I remember the transfer rate is locked but I don't remember why. I'm really busy right now so if you don't hear from me within a couple of weeks, ping me.

alza36sc105, 2023-04-07:
@morrownr 
Thank you very much for your comment.

closed 2023-04-07 by alza36sc105

alza36sc105, 2023-04-07:
Hi
The transfer rate can be changed.
I added the following code in the rtw_monitor_xmit_entry as a test and tried to capture the Frames.
Frames were being transferred at 2Mbps.
I injected the CTS frame as a test.

I'm planning on testing some other things. For example, can it be transferred at 40Mhz Width...

[image]
[image]

ZerBea, 2023-10-01:
Just to cross reference this problem. Frame injection is not working.

morrownr, 2023-10-01:
@ZerBea 

Thanks for the info. I'll look into as a I can. It might be time to see if I can get a newer version the driver as this one is getting old... speaking of drivers. Have you checked to see if the rtw88_8822bu.ko driver is handling frame injection? I've been testing it for managed mode and ap mode but not monitor mode.

ZerBea, 2023-10-02:
@morrownr 
Thanks for the reply. No, I can't test that because I don't have such a device.
I only got a bug report and a pcapng dump file to analyze.
https://github.com/ZerBea/hcxtools/issues/316#issuecomment-1740791941

I've forwarded your recommendation:
https://github.com/ZerBea/hcxtools/issues/316#issuecomment-1742453064