morrownr/88x2bu-20210702 #142: (info) New tested device

williamdes, 2023-04-06:
I tested `Bus 002 Device 004: ID 2001:331e D-Link Corp. 802.11ac NIC` on Debian **12 (bookworm)**

```
root@helixia:~/src/88x2bu-20210702# ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 2/2 (in-use/total processing units)
: 1645176 (total system memory)
: 6.1.0-7-amd64 (kernel version)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
./dkms-make.sh..............................................................................................
Signing module /var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-7-amd64/updates/dkms/
depmod..........
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

Do you want to apply the new options by rebooting now? (recommended) [Y/n] n

```

And it works ! 🎉 

```
wlxa8637d2db002: connected to TP-Link_60B2 1
        "D-Link Wi-Fi"
        wifi (rtl88x2bu), A8:63:7D:2D:B0:02, hw, mtu 1500
        inet4 192.168.1.90/24
        route4 192.168.1.0/24 metric 601
        route4 default via 192.168.1.1 metric 601
        inet6 fe80::1c0a:4c0c:b04c:831/64
        route6 fe80::/64 metric 1024
```

Please also have a look to https://bugs.debian.org/1008295 where a similar project https://github.com/RinCat/RTL88x2BU-Linux-Driver was told that Linux 6.2 would support some more drivers. Maybe it's worth documenting it

Here is another similar project: https://github.com/cilynx/rtl88x2bu

williamdes, 2023-04-06:
Posted https://linux-hardware.org/?probe=5cb163c75a

morrownr, 2023-04-06:
Hi @williamdes 

> And it works ! tada

Of course it works.

>  ...was told that Linux 6.2 would support some more drivers.

I have been aware of the project to merge an in-kernel driver for about a year. It is now included in kernel 6.2. Performance is not that good right now but I am seeing patches go in that should help performance. This driver automatically blacklists the in-kernel driver. To reactivate the in-kernel driver, simply run `sudo sh remove-driver.sh`.

> Maybe it's worth documenting it.

It is. The Main Menu for this site is located at:

https://github.com/morrownr/USB-WiFi

Menu item 2 is a list of Plug and Play adapters. I have already included a section for the rtl8812bu chipset.

@morrownr

williamdes, 2023-04-06:
Do you want me to find a web page to link on the README this device or is it not useful?

Closing since you did see this info, it will help Google searches too
Thanks for the quick reply

williamdes, 2023-04-06:
Also, you can update the README to say Debian 12 is supported

morrownr, 2023-04-06:
> Also, you can update the README to say Debian 12 is supported

Got it. Thanks.