morrownr/88x2bu-20210702 #146: Raspberry Pi 400 Installation Error

jakobkarosseit, 2023-04-29:
``
"Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums."
``

morrownr, 2023-04-29:
That message is for you, not me. Head on over to the RasPiOS forums and seek an answer.

The error is not in this driver.

Regards

mumixam, 2023-04-30:
@jakobkarosseit I actually just compiled this a few minutes ago. The issue is you updated your kernel and have not rebooted so your headers version is newer than the running kernel

jakobkarosseit, 2023-04-30:
@mumixam could you pls tell me what i have to do?

Feedy88, 2024-02-04:
@logicbreaks I don't know if your issue is persistent, but this fixed it for me: https://github.com/morrownr/88x2bu-20210702/issues/154