morrownr/88x2bu-20210702 #149: Add support for [Bus 001 Device 008: ID 056e:4011 Elecom Co., Ltd WDB-867DU3S]

Taniyan1015, 2023-05-17:
I appreciate this project because some PCs do not have Wi-Fi interface.

>>If you are sure that your adapter contains either a rtl8812bu
>>or rtl8822bu chipset and the ID is not shown above, please provide
>>the follow information so that I can include your adapter in the
>>driver source code so that the driver will work with your adapter:
>>
>>1. VendorID:DeviceID
>>2. Link to device on internet

I bought Wi-Fi(USB) adapter and check hardware ID.
It seems that this ID is changed from 0BDA:B812 to 056e:4011.  
>>Bus 001 Device 008: ID 056e:4011 Elecom Co., Ltd WDB-867DU3S

Therefore, I would like you to add 056e:4011 to your project.
I checked this adapter by windows PC, I think contoller may be RTL8812BU.

Thanks
Taniyan1015

Taniyan1015, 2023-05-18:
Dear @morrorwnr

I appreciate your supporting urgently.
I tried improved your code in following 2 conditions. 
As a result, my adapter(056e:4011) was able to communicate to Internet.

(1) 5.15.0-71-generic / lubuntu 20.04.6-LTS(Focal Fossa)
(2) 5.15.0-72-generic / lubuntu 20.04.2-LTS(Jammy Jellyfish)
 
Thank you very much for helping me !

morrownr, 2023-05-18:
> Thank you very much for helping me !

You are welcome.