morrownr/88x2bu-20210702 #150: RTL8832CU compatibility

aquilarubra, 2023-05-18:
I have a new RTL8832CU (https://es.aliexpress.com/item/1005005544770386.html?spm=a2g0o.productlist.main.1.78ff96c6qceFKn&algo_pvid=9ee2eb8d-9883-4709-90f5-0dc6d98c40da&algo_exp_id=9ee2eb8d-9883-4709-90f5-0dc6d98c40da-0&pdp_npi=3%40dis%21EUR%2161.26%2130.62%21%21%21%21%21%402145277316844042289821012d07b5%2112000033481930568%21sea%21IT%211918944861&curPageLogUid=OMcZ7AW58cDB)

It's a very affordable tri band usb wifi card. I don't know if this driver could support it, but I am available to test. I am on Arch Linux, kernel 6.3.2.

Once inserted, it gives:
0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

After an eject cdrom:
0bda:c832 Realtek Semiconductor Corp. 802.11ax WLAN Adapter

morrownr, 2023-05-18:
Good day @aquilarubra 

Let me see if I can help a fellow Linux user out.

> It's a very affordable tri band usb wifi card. 

Look at the adapter in the link below. It is more affordable and it is Plug and Play. The driver was in the Linux kernel last year before we could buy adapters.

https://www.aliexpress.us/item/3256804283254522.html?gatewayAdapt=glo2usa

> I don't know if this driver could support it...

This driver cannot support that adapter. If a Linux driver is available for this Realtek chipset, I am not aware of it. 

As far as WiFi 6 goes, Realtek orginally came out with the rtl8852/32au chipset a couple of years ago. They initially released an out-of-kernel driver but have never offered any updates. It was a terrible driver. I did my best to get it working well enough to make it available here but in the end, I had to give up. It appears Realtek has stopped production on that chipset so that chipset will likely end up being abandonware. I am currently working on a driver for the rtl8852/32bu. It is going slowly but at least there is a driver.

This rtl8852/32cu appears to be the first WiFi 6e chipset from Realtek. Mediatek is producing two WiFi 6e (the e means tri-band, 6 GHz) chipsets. The mt7921 and my7922. The only difference in the chipsets is the mt7922 support 160 MHz channel width whereas the mt7921 supports 80 MHz channel width. Both chipsets use the same driver that is in the kernel and is an excellent driver. If you were to buy the CF-953AX adapter in the link I added above, you would simply plug the adapter in and it would work on the latest version of Arch. It has been plug and play on Ubuntu since the 22.10 version. It is plug and play on the latest release of the Raspberry Pi OS.

Let me point you to some additional resources:

The Main Menu to this site is: https://github.com/morrownr/USB-WiFi

The Main Menu provides a lot of information to Linux users about USB WiFi adapters.

Menu item 1 gives some good background on the USB WiFi adapter market. Recommend you read it.

Menu item 2 provides a LONG list of adapters that are plug and play on Linux as well as Links to products and reviews.

We use Issues to ask questions, post news and information and discuss things. Please feel free to ask some questions.

My recommendation is for you to consider sending your adapter back for your money back as it will likely be a long, frustrating journey if you try to make it work on Linux. I hope to see you in Issues at the Main Menu.

Regards,

@morrownr

aquilarubra, 2023-05-18:
Thank you, I saw the Comfast. However, it is not tri band.
Practically, only Netgear A8000 is tri band and compatible with Linux, with availability on the market. And it costs over three times more. Hopefully, the RTL8832CU will be considered some day. I will try to ask in the linux-wireless list.

morrownr, 2023-05-18:
> I saw the Comfast. However, it is not tri band.

Yes it is. Numerous Linux users have reported that it is tri-band. I have the CF-951AX which uses the same chipset. It is tri-band. Mediatek has not made a WiFi 6 chipset that is NOT tri-band. The Windows driver may be limited to AX1800 but the Linux driver supports the full tri-band capability.

Some sellers of adapters with the mt7921au chipset are only advertising as AX1800 but we really do not know why other than when they started production Windows did not support 6 GHz.

> Practically, only Netgear A8000 is tri band and compatible with Linux, with availability on the market. 

Okay, so, you did not read the links I provided. I'm trying to help you. 

The ALFA AXML also advertises as AXE3000 tri-band. Both the ALFA and Netgear A8000 use the same exact chipset and use the same driver as the Comfast CF-953AX.

> I will try to ask in the linux-wireless list.

You may not get an answer at linux-wireless as that is not what that list is for.

Regards

aquilarubra, 2023-05-18:
Thank you again. Now I understand better and found further references.

nikkopt, 2024-09-04:
> Look at the adapter in the link below. It is more affordable and it is Plug and Play. The driver was in the Linux kernel last year before we could buy adapters.
> 
> https://www.aliexpress.us/item/3256804283254522.html?gatewayAdapt=glo2usa

Unfortunately that link now goes to the version 2 of the adapter which uses a Realtek chip

morrownr, 2024-09-04:
Hi @nikkopt 

Thanks for the info. I deleted that msg with the link.

To always get the current recommendations, see the Plug and Play List. It is menu item 2 on the Main Menu:

https://github.com/morrownr/USB-WiFi

There are currently 6 WiFi 6 adapters in the mt7921au section but there are many adapters in the Plug and Play List depending on what you need.

incyi, 2025-04-13:
I have this wireless card:
https://nl.aliexpress.com/item/1005006167646349.html

Shows up as:
0bda:c832 Realtek Semiconductor Corp. 802.11ax WLAN Adapter

I'm running Debian 12 with 6.1.0-31-amd64 kernel.

Nothing works.