morrownr/88x2bu-20210702 #151: Ur my last hope

Anakonder98, 2023-05-18:
Hii, today i follow your guide to install my usb external wifi card: https://www.amazon.it/dp/B0BHNKSJBM?psc=1&ref=ppx_yo2ov_dt_b_product_details
for virtualbox machine. The usb work really great in another laptop (with kali iso installed without other system os), i try to configure it in a virtual machine and then all the problem starts here. At the finish of all things i have the menu with "wifi" to search nearby connection but doesn't connect to mine or other, it try forever. Idk if i need to check the config system and maybe i miss something in there (i need to delete the # when i want the line is readed right) i pray you if u can help me to resolve this pain. I can't use my laptop to make all the things i need to do, i want make it run on a vm for a better "safe place" If u need something like how i set the config file let me know it. 

*i try with airodump to start the wlan0mon, it seems activate but when i try to scan it doesn't appear as 0mon and i can't scan
ifconfig: https://ibb.co/3r14JL0

morrownr, 2023-05-19:
Hi @Anakonder98 

> i try to configure it in a virtual machine and then all the problem starts here. 

You did not invent problems with VM's. It is common.

I'm probably the last person you want helping you with a VM. I use dual boot if I need more than one distro of Linux on a system. I don't do Windows. Never have. I do know a lot about wifi. I can try to help but you are not providing much info that could help. Hardware, distro versions, results of:

$ iw dev

Anakonder98, 2023-05-19:
> 

https://ibb.co/cXVwx2B
https://ibb.co/gVQWGnV

this is the vm machine and the command. I use the ISO on the official site version 2023.1

Anakonder98, 2023-05-19:
I read and i follow all the steps, https://ibb.co/gWVqcdy 
this is the setup where i cannot see the filter create it
and when i log in into the machine and try 
https://ibb.co/ysGD3H1
this is the result

Anakonder98, 2023-05-19:
> 

ill put my config file as text maybe there is something wrong with that

# /etc/modprobe.d/88x2bu.conf
#
# Purpose: Allow easy access to specific driver options.
#
# Note:	pull-request: wireless-next-2022-12-12
#		wireless-next patches for v6.2
#		wifi: rtw88: Add rtw8822bu chipset support
#
# The following line blacklists (deactivates) the above in-kernel driver.
blacklist rtw88_8822bu
#
# Edit the following line to change, add or delete options:
options 88x2bu rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_switch_usb_mode=0
#
# Note: To activate USB3 mode, change rtw_switch_usb_mode above to rtw_switch_usb_mode=1
#
# Note: The above `options` line is a good default for managed mode. Below is
# an example for AP mode. Modify as required after reading the documentation:
#options 88x2bu rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=1 rtw_switch_usb_mode=1 rtw_dfs_region_domain=1
#
# After editing is complete, save this file (if using nano: Ctrl + x, y, Enter)
# and reboot to activate the changes.
#
# Documentation:
#
# -----
#
# Log options ( rtw_drv_log_level )
#
# 0 = NONE
# 1 = ALWAYS (default)
# 2 = ERROR
# 3 = WARNING
# 4 = INFO
# 5 = DEBUG
# 6 = MAX
#
# Note: You can save a log file that only includes RTW log entries by running
# the following in a terminal:
#
# sudo ./save-log.sh
#
# Note: The name of the log file will be `rtw.log`.
#
# -----
#
# LED options ( rtw_led_ctrl )
#
# 0 = Always off
# 1 = Normal blink (default)
# 2 = Always on
#
# -----
#
# VHT options ( rtw_vht_enable )
#
#  0 = Disable
#  1 = Enable (default)
#  2 = Force auto enable (use only for 5 GHz AP mode)
#
# Notes:
# - A non-default setting can degrade performance greatly in managed mode.
# - Option 2 allows 80 MHz channel width for 5GHz AP mode, such as when
#   you are using hostapd.
#
# -----
#
# Power options ( rtw_power_mgnt )
#
# 0 = Disable power saving
# 1 = Power saving on, minPS (default)
# 2 = Power saving on, maxPS (not recommended for AP mode)
#
# -----
#
# Beamforming options ( rtw_beamform_cap )
#
# 1 = SU Beamformer (recommended for AP mode)
# 2 = SU Beamformee
# 3 = SU Beamformer and SU Beamformee
# 10= SU Beamformee and MU Beamformee (default)
# 11= SU Beamformer and SU Beamformee and MU Beamformee
#
# Note: MU Beamformer is not supported.
#
# From the source code:
#
# /* /os_dep/linux/os_intfs.c
#  *
#  * BIT0: Enable VHT SU Beamformer
#  * BIT1: Enable VHT SU Beamformee
#  * BIT2: Enable VHT MU Beamformer, depend on VHT SU Beamformer (not supported)
#  * BIT3: Enable VHT MU Beamformee, depend on VHT SU Beamformee
#  */
#
# -----
#
# USB options: ( rtw_switch_usb_mode )
#
# 0 = No switch (default)
# 1 = Switch from usb 2.0 to usb 3.0
# 2 = Switch from usb 3.0 to usb 2.0
#
# -----
#
# DFS Options ( rtw_dfs_region_domain )
#
# 0 = NONE (default)
# 1 = FCC
# 2 = MKK
# 3 = ETSI
#
# Notes:
# - Activates DFS channels in AP mode.
# - DFS FCC 80 MHz channels for hostapd: 52(58), 100(106), 116(122) and 132(138)
# - For more information: https://en.wikipedia.org/wiki/List_of_WLAN_channels
#
# Note: An AP needs to listen on a DFS channel for a period of 60 seconds
# before transmitting on the channel. If any radar pulses are detected,
# the AP cannot use that channel and will have to try a different channel.
#
# -----
#
# Wireless Mode options ( rtw_wireless_mode )
#
# 1  = 2.4GHz 802.11b
# 2  = 2.4GHz 802.11g
# 3  = 2.4GHz 802.11b/g
# 4  = 5GHz 802.11a
# 8  = 2.4Hz 802.11n
# 11 = 2.4GHz 802.11b/g/n
# 16 = 5GHz 802.11n
# 20 = 5GHz 802.11a/n
# 64 = 5GHz 802.11ac
# 84 = 5GHz 802.11a/n/ac
# 95 = 2.4GHz 802.11b/g/n 5GHz 802.11a/n/ac (default)
#
# -----
#
# Country Code options ( rtw_country_code )
#
# Note: Allows the Country Code to be set in cases where it is unable to
# be obtained otherwise.
#
# URL: http://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
#
# Example for the US: rtw_country_code=US
# Example for Panama: rtw_country_code=PA
# Example for Norway: rtw_country_code=NO
# Example for Kuwait: rtw_country_code=KW
# Example for Taiwan: rtw_country_code=TW
#
# -----
#
# To see all options that are available:
#
# $ ls /sys/module/88x2bu/parameters/
#
# -----
#
# To see the values that are in use:
#
# $ grep [[:alnum:]] /sys/module/88x2bu/parameters/*
#
# -----
#
# hostapd setup information for rtl88x2bu
# Note: The best settings can vary but the following may be a good place to start.
#
# /etc/modprobe.d/88x2bu.conf
# options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1
#
# Note: The best setting for `rtw_dfs_region_domain=` will depend on your location.
#
# /etc/hostapd/hostapd.conf
#
# hw ht capab: 0x963
# ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
#
# hw vht capab: 0x03c001a2
# vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# -----

morrownr, 2023-05-19:
Thanks for providing the information but I do not have adequate experience with VM's to be much help. Let me suggest a couple of things:

- Try going to the Kali forums and post a help request. Many Kali users use VM's.

- Try dual booting: https://www.kali.org/docs/installation/dual-boot-kali-with-windows/

I wish I could be more help.

Anakonder98, 2023-05-19:
> Thanks for providing the information but I do not have adequate experience with VM's to be much help. Let me suggest a couple of things:
>
> * Try going to the Kali forums and post a help request. Many [...]

no worries, is a good thing that u reply and try to help me. Not all people do that and for this thank you so much for doing it