morrownr/88x2bu-20210702 #153: TP-Link AC1300 USB wireless device not running at full speed under hostapd

lsdijk, 2023-05-21:
I recently got a TP-Link AC1300 USB wireless device that I am planning to use as a hot spot - in essence, I will plug it into my Slackware 15.0 desktop (with a 5.15.27 kernel) and then will let hostapd do its magic.

In a nutshell, after building and installing the appropriate kernel module available at this web site, this setup works, with a wrinkle: when I use it to connect to an external 5 GHz network the device attains a bit rate of 867 Mb/s (as reported by iwconfig) but when I use it as a hot spot with hostapd the maximum bit rate is just 174 Mb/s.

In detail, when using Network Manager to connect through this device to a 5 GHz network named A_5G_Network, the output from iwconfig is the following:

# iwconfig
lo no wireless extensions.

eth0 no wireless extensions.

tun0 no wireless extensions.

wlan1 IEEE 802.11AC ESSID:"A_5G_Network" Nickname:"<WIFI@REALTEK>"
Mode:Managed Frequency:5.22 GHz Access Point: 6C:CD:D6:10:DC:A3
Bit Rate:867 Mb/s Sensitivity:0/0
Retry:off RTS thr:off Fragment thr:off
Encryption key:****-****-****-****-****-****-****-**** Security mode:open
Power Management:off
Link Quality=46/100 Signal level=45/100 Noise level=0/100
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

When configured as a hot spot with hostapd, the output from iwconfig is now the following:

# iwconfig
lo no wireless extensions.

eth0 no wireless extensions.

tun0 no wireless extensions.

wlan1 IEEE 802.11bgn ESSID:"MyHotSpot" Nickname:"<WIFI@REALTEK>"
Mode:Master Frequency:5.765 GHz Access Point: 30:DE:4B:75:7D:59
Bit Rate:174 Mb/s Sensitivity:0/0
Retry:off RTS thr:off Fragment thr:off
Encryption key:off
Power Management:off
Link Quality=2/100 Signal level=2/100 Noise level=0/100
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

The hot spot itself works - I can connect to it from my phone, and I get Internet access all right. But, at this reduced speed. The contents of my hostapd.conf file are the following:

interface=wlan1
driver=nl80211
hw_mode=a
channel=153
ieee80211d=1
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
country_code=US

ssid=MyHotSpot
wpa=3
wpa_psk_file=/etc/hostapd/hostapd.wpa_psk
wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256
wpa_pairwise=TKIP
rsn_pairwise=CCMP
wpa_group_rekey=600
wpa_gmk_rekey=86400
wpa_ptk_rekey=600
ctrl_interface=/var/run/hostapd

The contents of my /etc/modprobe.d/88x2bu.conf file are

blacklist rtw88_8822bu
options 88x2bu rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=1 rtw_switch_usb_mode=2 rtw_dfs_region_domain=1

Intriguingly, if I load the module and attempt to start hostapd immediately afterward, the bit rate reported is 144.4 Mb/s, rather than 174 Mb/s. The latter I only get after I have used the device to connect to some external 5 GHz network - which, as I said above, it is able to do at 867 Mb/s. After that, when launching hostapd the bit rate is 174 Mb/s.

I don't know whether what I am reporting is a limitation in the kernel module itself or in hostapd, or whether hostapd can be configured so that the device will attain the full 867 Mb/s bit rate.

morrownr, 2023-06-02:
Hi @lsdijk 

Sorry for the delay. I have been on the road.

I see some problems:

> wpa=3
wpa=2

> rtw_switch_usb_mode=2 (this setting forces usb2)
rtw_switch_usb_mode=1

lsdijk, 2023-06-02:
Thanks. I set wpa=2 in hostapd.conf, and rtw_switch_usb_mode=1 in 88x2bu.conf. It seems to have made no difference: after stopping hostapd and launching it again with hostapd -t /etc/hostapd/hostapd.conf, the output from iwconfig is the same as described above.

morrownr, 2023-06-06:
You might want to look at item 9, AP Mode, in the Main Menu:

https://github.com/morrownr/USB-WiFi

While you may not be creating an AP exactly like this guide outlines, the guide does have a LOT of information that may help you.

lsdijk, 2023-06-06:
Thanks, that looks like a potentially useful source of relevant information. I will most certainly check it out.

 
 

https://github.com/morrownr/USB-WiFi