morrownr/88x2bu-20210702 #154: Fix for kernel/headers mismatch on Rasberry Pi OS

H179922, 2023-05-28:
I'm sure there is a better way to get this message to you but I'm a github luddite. 

I've been trying to install these drivers for a techkey and went through the header troubleshooting mismatch guides here and basically all over the internet. So this fix worked for 88x2bu and 8821au for me. 

I found a fix finally, and it was a simple edit to the boot config so the 64-bit drivers don't load presumably. 

`Can you try adding "arm_64bit=0" to config.txt, reboot and try again?`
source: https://forums.raspberrypi.com/viewtopic.php?t=350269

that's it. added that comment to the config.txt and rebooted and was able to successfully install the drivers you provided for both chipsets. 

hopefully this can be helpful to others that may run into this issue.

morrownr, 2023-05-31:
Thank you so much. I knew this was a 32 bit problem as I run the 64 bit version on my Pi3B and Pi4B and I had not seen it. I have been too busy to dig out a solution. I marked this issue as "good info" so others may read and learn.

Thanks again.

tester1969, 2023-06-05:
After upgrade to 6.1+ kernel same issues. Besides the 32 bit I had to give also the following:
sudo ln -s /usr/src/linux-headers-6.1.21-v7l+/ /lib/modules//6.1.25-v7l+/build
because "build" directory was missing. Also dkms failed with error 7 and had to put the module manually