morrownr/88x2bu-20210702 #155: 8812bu ok - worked.

phk68, 2023-05-31:
I have xubuntu 22.04 jammy. I had many problems to install. I follow the steps again and now works fine. Than you very much.

cristian-ciobanu, 2023-06-05:
@morrownr so in kernel 6.3 and newer support for 88x2bu is stable enough and can be used without installing this out-of-kernel driver from this Github repo ?

Do you know also if AP support for this chipset is implemented in kernel 6.3 and running fine ?

Thank you

cristian-ciobanu, 2023-06-06:
Hi @morrownr 
Thanks for replying.

Right now I'm using the driver from this repository and I have an TP-Link Archer T3U Plus which I use it in AP mode. I had some problems with this driver since very often the SSID was disappearing and every time I needed to re-plug the device into the USB port and restart hostapd. Recently I found some information about this problem on another issue reported for this 88x2bu driver which recommended to disable power saving. So far in the last few days the connection seems stable.
That is one reason I asked if the in-kernel driver is working fine so I can switch and use it instead of this out-of-kernel driver.