morrownr/88x2bu-20210702 #159: Installation problem, (bad exit status: 2) (log is given) (Rpi Zero W2)

YasinBedirhanSimsek, 2023-06-28:
I have a Raspberry Pi Zero W2 and my USB wifi dongle is "Everest ewn-ac1200"

I need a 5.0GHz connection between my PC and the Raspberry Pi for the SSH and the 2.4GHz is not usable in my case. 

I can't the install driver because I keep getting "bad exit status:2" error during installation with install-driver.sh

```
sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get
Bus 001 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
8812au, 5.6.4.2_35491.20191025, 6.1.21-v8+, aarch64: installed (WARNING! Diff between built and installed module!)
rtl88x2bu, 5.13.1: added
rtl88x2bu, 5.8.7.1: added
```

```
sudo uname -a
Linux  <deviceName>6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

install-driver.sh output:

```
sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 2/4 (in-use/total processing units)
: 429400 (total system memory)
: 6.1.21-v8+ (kernel version)
: gcc (Debian 10.2.1-6) 10.2.1 20210110
: dkms:2.8
EFI variables are not supported on this system
: ---------------------------

Checking for previously installed drivers.
rtl88x2bu, 5.8.7.1: added
: ---------------------------

Removing a dkms installation.
Error! The module/version combo: rtl88x2bu-5.13.1
is not located in the DKMS tree.
Removing 88x2bu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl88x2bu-5.13.1
Removal complete.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
                 /usr/src/rtl88x2bu-5.13.1

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area..........
./dkms-make.sh..........................................................................................................................................................................................................................................................................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v8+ (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

```
*******************************************************************************************

Error section in the log file

```
DKMS make.log for rtl88x2bu-5.13.1 for kernel 6.1.21-v8+ (aarch64)
Wed 28 Jun 07:14:36 +03 2023
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/home/enveko/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ioctl_query.o
....
....
.... There are lots of lines here like the ones above and below
....
....
 CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/halrf/halrf_psd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/halrf/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/halrf/rtl8822b/halrf_rfk_init_8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/btc/halbtc8822bwifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/btc/halbtc8822b1ant.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/btc/halbtc8822b2ant.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/platform/platform_ops.o
  LD [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.o
  MODPOST /var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers
ERROR: modpost: "__arch_copy_to_user" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "alt_cb_patch_nops" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__const_udelay" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:126: /var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers] Error 1
make[1]: *** [Makefile:1964: modpost] Error 2
make[1]: Leaving directory '/home/enveko/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
make: *** [Makefile:2523: modules] Error 2
```

morrownr, 2023-06-28:
Hi @YasinBedirhanSimsek 

I have updated the installation script. Run:

$ git pull
$ sudo sh install-driver.sh

Please post the onscreen log.

YasinBedirhanSimsek, 2023-06-28:
@morrownr Thank you for your response but I somehow managed to install it and connect to my 5GHz network. It is working fine for now.

I just reinstalled the 5.15 64-bit Raspberry OS Lite with Raspberry Pi Imager again and followed the steps on the readme section. 

**BUT**

I don't know what fixed it. I might have cloned the repo after you posted your fix. 

Again, thank you for your time.

YasinBedirhanSimsek, 2023-06-28:
I checked the script version on my cloned repo files and it was 

`SCRIPT_VERSION="20230628"`

So probably it was what you did and not the old kernel setup.

Anyway, if anyone else comes across this, you might as well give "old kernel trick" a try.

morrownr, 2023-06-28:
> So probably it was what you did and not the old kernel setup.

The main thing I did when I saw you had a multi core cpu but only a half gig of ram was told the script to pull back to using only 1 core  in that situation. That very well could be what fixed the problem. I'm sure the compile took a long time but you don't have the ram to handle more than one core.

YasinBedirhanSimsek, 2023-06-28:
Yes, you are right. Yesterday some other wifi adapter driver repo took me more than 10+hours to build but ended up failed. 

The swapfile size was 4gb but I guess it was not the enough or fast enough or suitable as a memory.

closed 2023-06-28 by YasinBedirhanSimsek

morrownr, 2023-06-28:
> Yes, you are right. Yesterday some other wifi adapter driver repo took me more than 10+hours to build but ended up failed.

That is one of the problems with the out-of-kernel drivers based on older technology. I realize that the internet does not get cleaned up so there is a problem with old information. The adapter you are using is supported with an in-kernel driver starting with kernel 6.2 so the RasPiOS is not there yet unless you want to replace the kernel with a newer one. I have a guide if you are interested.

Also, my main effort to help inform Linux users is at this url:

https://github.com/morrownr/USB-WiFi

Main Menu item 2 gives a long list of adapters that use in-kernel drivers that are modern Linux WiFi standards compliant. We use Issues to discuss things.

I did not answer one of your questions. The code in the dkms-make.sh file that changes how many codes with be used is as follows:

# avoid Out of Memory condition in low-RAM systems by limiting core usage
if [ "$sproc" -gt 1 ]; then
	if [ "$SMEM" -lt 1400000 ]; then
		sproc=2
	fi
	if [ "$SMEM" -lt 700000 ]; then
		sproc=1
	fi
fi