morrownr/88x2bu-20210702 #162: can it work with 8832bu?

dibu28, 2023-07-08:
can it work with 8832bu?

morrownr, 2023-07-08:
No.

Do you have a 8832bu? If so, I will add you to a private repo and you can help test. I am working on a 8832bu driver.

If you do not have a 8832bu, recommend you do NOT buy one. The driver I am working on is a challenge and I cannot guarantee it will ever be a fully serviceable driver.

dibu28, 2023-07-10:
Already bought it, in Turkey it was the cheapest option))

Everest AX1800 EWA1800 model. RTL8832BU - chipset written on package. And shows in Windows.

It was a problem to fing drivers even in Windows. And with drivers it is still a little bit laggy))

Yes, I can help you with testing.

Which wi-fi AX adapters do you recommend to buy for linux? Which are well supported?

morrownr, 2023-07-10:
Hi @dibu28 

> Yes, I can help you with testing.

Great. You will be getting an email that will help you get into the private repo. See you there.

> Which wi-fi AX adapters do you recommend to buy for linux? Which are well supported?

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

My site Main Menu is located at:

https://github.com/morrownr/USB-WiFi

birdofprey, 2023-08-16:
os: debian Gnu/Linux 12
kernel: Linux localhost 6.4.0-0.deb12.2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.4.4-3~bpo12+1 (2023-08-08) x86_64 GNU/Linux
dkms status: rtl88x2bu/5.13.1, 6.4.0-0.deb12.2-amd64, x86_64: installed
very very good

nilson7000, 2023-11-17:
Hello greetings, I bought CF-966AX with the RTL8832BU chipset, I would like to know what I can do to make it work?

morrownr, 2023-11-17:
Hi @nilson7000 

This site has a Main Menu to help users find the various things they may be looking for:

https://github.com/morrownr/USB-WiFi

I'd recommeding reading menu items 1 and 2 plus, in your case, see menu item 4 which includes a link to the rtl8832bu driver that is located at this site. The driver works reasonable well. I did the best I could with it. Let me know it goes.

@morrownr