morrownr/88x2bu-20210702 #164: Works in Fedora 37, kernel 6.3.12-100.fc37.x86_64, TP-Link Archer T3U

aki-k, 2023-07-13:
ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]

aki-k, 2023-07-29:
Keeps on working in Fedora 37, kernel 6.4.6-100.fc37.x86_64
Great work, thanks!

morrownr, 2023-07-30:
Thanks for the report @aki-k 

That report on the 6.4.6 kernel saved me some work. Anytime you have some extra time and would like to do some testing, stop by and post a msg.

Also, have you tried the in-kernel driver for this chipset?

aki-k, 2023-07-30:
I didn't even know it was included in the kernel.

morrownr, 2023-07-30:
> The TP-Link USB stick's green activity led is not blinking with rtw_8822bu...

I'd have to research the issue to see why that is but I am pretty sure is intentional. In fact, Alfa's last 2 usb wifi adapters that were built with Linux support in mind change the LED from a bright blinking light to a carefully placed dim LED that is on solid when the adapter is on and is off when no power is going to the adapter.

Keep in mind that the rtw_8822bu (in-kernel) driver and the rtl88x2bu (out-of-kernel) driver are very different.  The in-kernel driver you are using now is Linux Wireless Standards compliant (or it would not be in the kernel.) The in-kernel drivers are designed for regular desktop and server distros. The out-of-kernels drivers are designed for programmers working on embedded systems or industrial projects where you generally see locked distros with locked kernels. That is, no updates. Unfortunately this programmer oriented, out-of-kernel driver causes all kinds of problems for us mortals.

When I started this site a few years ago, I was getting questions about how to find adapters that work well and are easy to maintain on Linux. The result:

Go to the site Main Menu and select menu item 2:

https://github.com/morrownr/USB-WiFi

If you have any questions, let me know.

@morrownr

morrownr, 2023-07-30:
> |__ Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M

Are you using a USB2 port? If so, try a USB3 port and check again.

aki-k, 2023-07-30:
Yes I have the Archer T3U stick in a USB 2.0 port but it works well in it

aki-k, 2023-08-11:
Fedora 37 got the kernel update to 6.4.9-100.fc37.x86_64. Still working!

morrownr, 2023-08-12:
Sounds good. Have you tried the in-kernel driver with the adapter in a USB3 port to see if USB3 is working?

aki-k, 2023-08-12:
No but I'll try it later today

aki-k, 2023-08-13:
The TP-Link Archer T3U USB stick works in a USB3 port with rtw88_8822bu.

lsusb -t shows the speed as 480M? The port has a blue connector and a USB3 storage stick works fast in it.

```
$ lsmod | grep rtw
rtw88_8822bu           12288  0
rtw88_usb              28672  1 rtw88_8822bu
rtw88_8822b           229376  1 rtw88_8822bu
rtw88_core            245760  2 rtw88_usb,rtw88_8822b
mac80211             1527808  2 rtw88_core,rtw88_usb
cfg80211             1286144  2 rtw88_core,mac80211
```
```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M
    |__ Port 2: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 2: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 5: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 5: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 8: Dev 6, If 0, Class=Vendor Specific Class, Driver=rtsx_usb, 480M
```
```
$ iw dev wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr d0:37:45:47:23:05
        ssid ssid
        type managed
        wiphy 0
        channel 64 (5320 MHz), width: 80 MHz, center1: 5290 MHz
        txpower 20.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0
```

aki-k, 2023-08-13:
The green led of the T3U blinks in Windows 10 though (I have Windows 10 on dual-boot on the same laptop)

morrownr, 2023-08-13:
> lsusb -t shows the speed as 480M? 

480M means USB2. I'm not sure why it is not USB3.

> The green led of the T3U blinks in Windows 10 though (I have Windows 10 on dual-boot on the same laptop)

I'll see if I can get some answers about why the LED is not supported. In fairness to the in-kernel driver, it supports a lot of things the out-of-kernel driver does not.