morrownr/88x2bu-20210702 #167: Module not loaded kernel 6.4.6

ClodoaldoPinto, 2023-07-25:
```
# ll /lib/modules/6.4.6-200.fc38.x86_64/kernel/drivers/net/wireless/realtek/rtw88
total 560
-rw-r--r--. 1 root root  15720 Jul 23 21:00 rtw88_8723de.ko.xz
-rw-r--r--. 1 root root  33240 Jul 23 21:00 rtw88_8723d.ko.xz
-rw-r--r--. 1 root root  15704 Jul 23 21:00 rtw88_8723du.ko.xz
-rw-r--r--. 1 root root  15712 Jul 23 21:00 rtw88_8821ce.ko.xz
-rw-r--r--. 1 root root  34172 Jul 23 21:00 rtw88_8821c.ko.xz
-rw-r--r--. 1 root root  15884 Jul 23 21:00 rtw88_8821cs.ko.xz
-rw-r--r--. 1 root root  15716 Jul 23 21:00 rtw88_8821cu.ko.xz
-rw-r--r--. 1 root root  15680 Jul 23 21:00 rtw88_8822be.ko.xz
-rw-r--r--. 1 root root  40892 Jul 23 21:00 rtw88_8822b.ko.xz
-rw-r--r--. 1 root root  15908 Jul 23 21:00 rtw88_8822bs.ko.xz
-rw-r--r--. 1 root root  16044 Jul 23 21:00 rtw88_8822bu.ko.xz
-rw-r--r--. 1 root root  15732 Jul 23 21:00 rtw88_8822ce.ko.xz
-rw-r--r--. 1 root root  62576 Jul 23 21:00 rtw88_8822c.ko.xz
-rw-r--r--. 1 root root  15912 Jul 23 21:00 rtw88_8822cs.ko.xz
-rw-r--r--. 1 root root  15684 Jul 23 21:00 rtw88_8822cu.ko.xz
-rw-r--r--. 1 root root 112980 Jul 23 21:00 rtw88_core.ko.xz
-rw-r--r--. 1 root root  33796 Jul 23 21:00 rtw88_pci.ko.xz
-rw-r--r--. 1 root root  32304 Jul 23 21:00 rtw88_sdio.ko.xz
-rw-r--r--. 1 root root  27484 Jul 23 21:00 rtw88_usb.ko.xz
```
```
# lsmod | grep rtw
```
[make.log](https://github.com/morrownr/88x2bu-20210702/files/12166269/make.log)

morrownr, 2023-07-26:
Hi @ClodoaldoPinto 

You are aware that the rtw88 directory shown above is not associated with the driver in this directory?

Let's check something first. Run and post the result of:

$ lsusb

ClodoaldoPinto, 2023-07-26:
> You are aware that the rtw88 directory shown above is not associated with the driver in this directory?
> 
No
> Let's check something first. Run and post the result of:
> 
> $ lsusb
```
# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

morrownr, 2023-07-26:
Hi again @ClodoaldoPinto  o Hola otra vez

> ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio

That is a usb bluetooth adapter. The driver is located:

https://github.com/torvalds/linux/blob/master/drivers/bluetooth/btusb.c

That is in the Linux kernel so it should just work barring the VID/PID not being in the driver but we can cross that bridge if necessary.

I am not seeing a usb wifi adapter in your system. Is your wifi an internal card? Can you clarify why you were looking in this repo?

For what it is worth, the Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

ClodoaldoPinto, 2023-07-26:
Now it appears after unplugging and plugging:
```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 009: ID 2717:ff40 Xiaomi Inc. Mi/Redmi series (MTP)
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 019: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 003: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 013: ID 048d:5702 Integrated Technology Express, Inc. RGB LED Controller
Bus 001 Device 017: ID 056a:037a Wacom Co., Ltd CTL-472 [One by Wacom (S)]
Bus 001 Device 016: ID 0d8c:0102 C-Media Electronics, Inc. CM106 Like Sound Device
Bus 001 Device 015: ID 045b:0209 Hitachi, Ltd 
Bus 001 Device 011: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
Bus 001 Device 007: ID 045b:0209 Hitachi, Ltd 
Bus 001 Device 014: ID 046d:c506 Logitech, Inc. MX700 Cordless Mouse Receiver
Bus 001 Device 012: ID 046d:c33a Logitech, Inc. G413 Gaming Keyboard
Bus 001 Device 008: ID 046d:c548 Logitech, Inc. Logi Bolt Receiver
Bus 001 Device 004: ID 4348:7048 WinChipHead CH545
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
[image]

ClodoaldoPinto, 2023-07-26:
When I upgraded from 6.3 to 6.4.4 (the first 6.4 from Fedora) I had a problem where it always took minutes after booting to recognize the adapter. Then I upgraded to 6.4.6 and it did not load at all. I will boot back to 6.4.4

ClodoaldoPinto, 2023-07-26:
When I upgraded from 6.3 to 6.4.4 (the first 6.4 from Fedora) I had a problem where it always took minutes after booting to recognize the adapter. Then I upgraded to 6.4.6 and it did not load at all. I will boot back to 6.4.4

Now after replugging it is also loading in 6.4.6, but again with a delay after booting. And I reboot into 6.3.12 and it is the same problem. 

I guess it is something I did when I upgraded. I suspect this:
```
mokutil --import /var/lib/dkms/mok.pub
```
I issued it when I started having installation problems without understanding the exact meaning

morrownr, 2023-07-26:
> I had a problem where it always took minutes after booting to recognize the adapter.

There is a problem with the adapter being recognized as we saw when you ran `lsusb`. Registering so that it is visible with `lsusb` has nothing to do with the driver.

Let me backup for a moment. Do you realize that you do not need the driver in this repo? That is correct. Kernel 6.4 has an in-kernel driver for this chipset. From your original post:

-rw-r--r--. 1 root root  16044 Jul 23 21:00 rtw88_8822bu.ko.xz

That is the in-kernel driver and it is pretty good.

In fact, I had to add code to this driver to blacklist the in-kernel driver. Recommend you run `$ sudo sh remove-driver.sh` and reboot so that the in-kernel driver is reactivated and see what happens.

If you have a little extra time, you could read menu items 1 and 2 at the site Main Menu as there is more information about how things work:

https://github.com/morrownr/USB-WiFi

ClodoaldoPinto, 2023-07-26:
I removed the driver and rebooted and the kernel module is loaded (I guess)
```
$ lsmod | grep rtw
rtw88_8822bu           12288  0
rtw88_usb              28672  1 rtw88_8822bu
rtw88_8822b           229376  1 rtw88_8822bu
rtw88_core            245760  2 rtw88_usb,rtw88_8822b
mac80211             1527808  2 rtw88_core,rtw88_usb
cfg80211             1282048  2 rtw88_core,mac80211
```
But it never connects and freezes the desktop while trying to do so.

morrownr, 2023-07-26:
>and the kernel module is loaded (I guess)

Yes, the in-kernel driver is loaded.

See what the following says:

$ sudo dmesg | grep rtw

ClodoaldoPinto, 2023-07-26:
```
# dmesg | grep rtw
[   25.138280] rtw_8822bu 1-6:1.0: Firmware version 27.2.0, H2C version 13
[   33.474782] usbcore: registered new interface driver rtw_8822bu
[   33.477966] rtw_8822bu 1-6:1.0 wlp2s0f0u6: renamed from wlan0
```

morrownr, 2023-07-26:
That makes it appear that the in-kernel driver is loading just fine. There are no errors shown at all. 

Something external to the driver is likely the problem. I have some chores to do but will try to help more later.

morrownr, 2023-07-27:
I have seen multi-state adapters cause that are similar.

How about you disable usb_modeswitch and see if the adapter shows up as a flash driver or cd-rom?

Go to the Main Menu, https://github.com/morrownr/USB-WiFi ,and go down to menu item 7. At the end of the document iis how to disable. Then reboot and see what you get.

ClodoaldoPinto, 2023-07-27:
Makes no difference
```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2717:ff40 Xiaomi Inc. Mi/Redmi series (MTP)
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 003: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 008: ID 048d:5702 Integrated Technology Express, Inc. RGB LED Controller
Bus 001 Device 014: ID 056a:037a Wacom Co., Ltd CTL-472 [One by Wacom (S)]
Bus 001 Device 013: ID 0d8c:0102 C-Media Electronics, Inc. CM106 Like Sound Device
Bus 001 Device 012: ID 045b:0209 Hitachi, Ltd 
Bus 001 Device 009: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
Bus 001 Device 006: ID 045b:0209 Hitachi, Ltd 
Bus 001 Device 011: ID 046d:c506 Logitech, Inc. MX700 Cordless Mouse Receiver
Bus 001 Device 010: ID 046d:c33a Logitech, Inc. G413 Gaming Keyboard
Bus 001 Device 007: ID 046d:c548 Logitech, Inc. Logi Bolt Receiver
Bus 001 Device 004: ID 4348:7048 WinChipHead CH545
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

morrownr, 2023-07-27:
I just read this thread again from the beginning. If I faced the same situation, I would do 2 things:

1. Read the forums for your distro to see if others are seeing something similar.
2. Backup any important files and do a clean installation.

I don't think this driver is the problem.

Here are some thoughts. I am not trying to tell you what to do but rather passing on lessons I have learned over the years:

For video, I make sure systems that I buy use Intel and AMD video chipsets. They are solid and their Linux support is excellent. This dev box I am on right now is 11 years old. I have systems that are of various ages with the oldest being from the late 90's up to one that is less than 2 years old. There is no nvidia hardware in site. nvidia does not do very well with Linux support so I avoid their products.

For WiFi: I use and recommend Mediatek, Qualcomm and Intel chipsets. Their Linux support is excellent. Of that group, only Mediatek supplies usb chipsets and the chipsets are supported with standards compliant in-kernel drivers. Even though I maintain some Realtek drivers, I make it clean that I do not recommend Linux users buy adapters that use Realtek chipsets because Realtek's Linux support is subpar. You can get more details with menu items 1 and 2 on the Main Menu. I posted the link earlier.

Regards

ClodoaldoPinto, 2023-07-31:
My phone started having connection problems with my pc so I tryied changing it from a USB 2 port to a 3 one and it worked so I tried the same with the wifi adapter and now it is working with no problems with both in and out of kernel drivers.

The in kernel has the inconvenience that I can't pass parameters to it. Or can I?

closed 2023-07-31 by ClodoaldoPinto

morrownr, 2023-07-31:
> The in kernel one has the inconvenience that I can't pass parameters to it. Or can I?

You can but I will have to see what is available. One of the main differences between the in-kernel drivers and the out-of-kernel drivers is that most configuration is automatic with the in-kernel drivers. It is designed that way. Example: If you want the adapter to run in USB2 mode, put the adapter is a USB2 port and in a USB3 port, it will run in USB3 mode. The driver detects and switches automatically. The fact that I can't remember which parameters are available should tell you something... it works.