morrownr/88x2bu-20210702 #169: impossible detect access point one plus nord 2 5g with GHZ

alex5956, 2023-08-02:
hello i can't detect my phone one plus nord 2 5 5GHZ with wifite 2 in 5GHZ, only in 2.4 GHZ
why ?could you help me ?
i did the git pull today

morrownr, 2023-08-02:
Hi @alex5956 

I am confused as to what the problem is. Are you having a problem with a usb wifi adapter?

If you look at the top of Issues for a pinned message, it will spell out what to provide.

alex5956, 2023-08-02:
no i have not probelm to detect antothers access points

morrownr, 2023-08-02:
I cannot help you if I do not understand what the problem is.

alex5956, 2023-08-02:
ok

spacemishka, 2023-12-31:
The problem is there are two kernel versions involved. I tried to compile this driver on my Xiaomi Mi 9T to use it with Nethunter.
Nethunter has as of today Kernel 6.5 but the kernel for Android 10 is 4.14. When I do 'make install' then I am getting an error that the sources of 4.14 can not be found.

spacemishka, 2023-12-31:
I would suggest to close this issue since the driver needs to be compiled for the Android Kernel

alex5956, 2023-12-31:
Yes i think too,try to compile with android , then go back with more infos/logs