morrownr/88x2bu-20210702 #172: Question: Any package/support for rtl8812bu chipset drivers for OWRT?

TMPJoker, 2023-08-27:


morrownr, 2023-08-27:
What is OWRT?

TMPJoker, 2023-08-28:
https://openwrt.org/

morrownr, 2023-08-28:
I've been working on updating the README and I 'll merge it soon. It addresses this question so I'll copy and paste:

Note: OpenWRT is not supported. OpenWRT provides drivers for USB WiFi
adapters. OpenWRT provided drivers include support for the MT7921au
(AXE3000), MT7612u (AC1200), MT7610u (AC600) chipsets. It is a challenge
to use Realtek based adapters that use out-of-kernel drivers with
OpenWRT so it is strongly advised to use the already supported chipsets.

More info is available at the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 may be of interest. Menu item 2 gives information and links to specific adapters for the chipsets mentioned above, Feel free to ask questions in issues at the Main Menu.

Hope this helps.