morrownr/88x2bu-20210702 #173: where is firmware for Edimax EW-7822UAD?

bdantas, 2023-08-29:
Hello, morrownr. Thank you for providing the driver for these devices. I'm on Tiny Core LInux 14 x86_64 (with linux 6.1.2), which is a little different from mainstream distros, but I was able to successfully compile 88x2bu.ko with the help of your build instructions.

The Edimax EW-7822UAD was recently purchased and I should have it very soon. I'd like to be prepared to use it when it arrives. I will need 88x2bu.ko plus a firmware file, right?

I've searched the internet quite a bit and cannot find the name of the firmware file (foo.bin) that this devices needs. Do you know the name of the file and where I can download it from?

Thank you very much for any guidance you can provide.

bdantas, 2023-08-29:
Hi, Nick. When I search Debian's repository I can find rtl8812aefw_wowlan.bin and rtl8812aefw.bin in the package "firmware-realtek". Is the firmware for this device one of these two files? 

I cannot find anything anywhere named "rtl8812bu.bin" (that would be too easy!).

morrownr, 2023-08-29:
Hi @bdantas 

If your adapter has the rtl8812bu chipset, you have options as this chipset is supported by this out-of-kernel driver but it is also supported by the rtw88 in-kernel driver with kernels 6.2 and later.

This driver contains its own firmware so adding firmware is not necessary. Once you are using kernel 6.2 or later, you will need to add the needed firmware file. Here is the link to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu item 7 gives information about firmware for modern in-kernel drivers. The detailed info is for Mediatek drivers but early in the document is a note that gives the location and info for the rtw88 in-kernel driver firmware. The specific file for the rtl8812bu driver is:

rtw8822b_fw.bin

My testing has shown that with kernels 6.4 and later, the in-kernel driver is pretty darn good.

This driver installs a file to blacklist the in-kernel driver so you need to use remove-driver.sh to get rid of the blacklist or get rid of it manually.

Let me know if you need anything else.

Nick

bdantas, 2023-08-29:
> it is also supported by the rtw88 in-kernel driver with kernels 6.2 and later.
 
I'm on kernel 6.1.2 so need the out-of-kernel driver for now.

> This driver contains its own firmware so adding firmware is not necessary.
 
Wow, I would never have guessed. I thought driver (kernel module) and firmware were always two separate things. I'm glad I asked.

You and your github page are an invaluable resource, best in the world for helpful and accurate information about these devices. You're the man! Thank you.

closed 2023-08-29 by bdantas

morrownr, 2023-08-29:
You are welcome. Stop by anytime.