morrownr/88x2bu-20210702 #174: Problem on 6.4 kernel

sjtmbg, 2023-08-29:
Built 88x2bu driver on kali using kernel 6.4.0-kali3-amd64 with these warnings:
 CC [M]  .../88x2bu-20210702/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
.../88x2bu-20210702/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.c: In function ‘_lok_one_shot_8822b’:
.../88x2bu-20210702/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.c:960:13: warning: enum constant in boolean context [-Wint-in-bool-context]
  CC [M]  .../88x2bu-20210702/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
.../88x2bu-20210702/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.c: In function ‘_lok_one_shot_8822b’:
.../88x2bu-20210702/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.c:960:13: warning: enum constant in boolean context [-Wint-in-bool-context]
  960 |         if (DBG_RF_IQK) {
      |             ^~~~~~~~~~

88x2bu.ko file was created and 'make install' gave no errors.  However, when I unplugged the usb wifi card and plugged it back in, the system completely froze.  Rebuilt the driver and same thing happened 2nd time.  I plugged the wifi card into a nearby laptop and it worked with no problems using 88x2bu

morrownr, 2023-08-30:
Hi @sjtmbg 

There is a pinned issue that you need to read and follow:

(Info) Problem Reporting Guidelines - Please Read.

sjtmbg, 2023-08-31:
This is on kali 2023.3 using kernel 6.4.0-kali3-amd64

This is no longer an issue because the install-driver.sh script successfully installs the driver using dkms.

I prefer using 'make clean; make; sudo make install' when I do a kernel upgrade because I feel I learn more that way.  I don't want to be treating this repository as a helpdesk - I want to learn more about how it works and perhaps even contribute in some small way.

Even after installing using the script, when I remove the driver from dkms using the remove_driver.sh script and then attempt to install using make/make install, I get the same system freeze.

I have included the install-driver.sh output below for completeness **but note that this is the output of a successful install.**

[88x2bu_build.txt](https://github.com/morrownr/88x2bu-20210702/files/12481361/88x2bu_build.txt)

I could add the output of lsusb, rfkill, etc, but that would be for the successful install only, obviously I can't run those commands after the system freezes.

morrownr, 2023-08-31:
Hi @sjtmbg 

A little info goes a long way. I see exactly what the problem is:

When you install with:

$ make clean
$ make
$ sudo make install

...you make a situation where 2 drivers for the same device are installed and active. Your version of Kali has an in-kernel driver that will automatically drive your adapter. When you do a manual installation, you have conflicting drivers.

My install-driver script blacklist the in-kernel driver and remove-driver.sh removes the backlist. If you want to do a manual installation, you will need to blacklist the in-kernel driver but I recommend against that as the in-kernel driver is actually the one that you probably want to use if using Kali as it should be more compatible with monitor mode.

> I want to learn more about how it works and perhaps even contribute in some small way.

If you are interested, I could use some help in a few places here at the site. I could certainly use a tester and dev for work on install-driver.sh and remove-driver.sh. I could also use help in my monitor mode repo. All of the scripts are script language. You would learn a lot and I would get some help.

Cheers

sjtmbg, 2023-08-31:
Of course!  I should have realized that -  sorry!

Yes, I'd like to help.  Please point me in the right direction to get started.

morrownr, 2023-08-31:
Be glad to have your help. This is about learning so this needs to be enjoyable.

Let me recommend you go to and study the contents of the site Main Menu first:

https://github.com/morrownr/USB-WiFi

Let me know when you are ready to discuss a project or two.

Nick

sjtmbg, 2023-08-31:
Thanks for the recommendation - will do.

One more comment  about the 6.4 kernel.  When  I blacklist the in-kernel driver and install 88x2bu, it works, no problem, so thanks again for that help.  I will note though that I've been running a 6.4 kernel on arch (now up to 6.4.11-arch2-1) for about two months and I never had this problem (I've never had a system freeze when installing the 88x2bu driver on any distro/kernel).  On arch, it will choose the in-kernel driver by default when you plug in the usb card and you can use the 88x2bu driver by blacklisting and rebooting.

sjtmbg, 2023-09-07:
> Be glad to have your help. This is about learning so this needs to be enjoyable.
> 
> Let me recommend you go to and study the contents of the site Main Menu first:
> 
> https://github.com/morrownr/USB-WiFi
> 
> Let me know when you are ready to discuss a project or two.
> 
> Nick

I've done the recommended reading - ready to discuss...

morrownr, 2023-09-07:
Here is my non-public email address. Please contact me:

morrownr@gmail.com

closed 2023-09-07 by morrownr